orgua/OneWireHub #1: Pro Micro Board compatability

GraemeTownsend, 2016-03-09:
Hi,

I've just come across your project and I'm impressed with the work. I'm trying to get it to run with my existing project for a test.

I have a LEONARDO board running as a master One Wire device. Connected are two DS18B20 temperature probes that are detected at start up, and polled for temperature with no issue.

I have compiled the OneWireHub example on a Pro Micro (LEONARDO clone, ATmega32) and linked it in as a Slave device. I cannot detect it and get no reading back, however, the debug output shows that it is receiving commands, (example below.)

Is the Pro Micro compatible? Is there something else I should be doing or another example I should try?

Thanks.
Graeme.
 

Example Error debug output:
Err: incorrect onewire command
cmd : 0   HEX: 0
Err: incorrect onewire command
cmd : 0   HEX: 0
Err: incorrect onewire command
cmd : 204   HEX: CC
cmd : 85   HEX: 55
cmd : 0   HEX: 0
Err: incorrect onewire command

GraemeTownsend, 2016-03-10:
Hi,

Done some more experimenting.
Using your "debug-master_DS18x20" on a LEONARDO. Works fine. I Can see real DS18B20's on the OneWire no problem.

Using your "Sensor_DS18B20" on a Pro Micro board, I get errors. Constant stream of;
   Err: read timeslot timeout
   cmd : 240   HEX: F0
And the Master pumps out;
   No more addresses.

If I attach a real DS18B20 the Master see's it instantly and starts to display the value. The Slave then starts displaying the below but the master never Id's the Slave:

   cmd : 240   HEX: F0
   cmd : 240   HEX: F0
   cmd : 85   HEX: 55
   cmd : 85   HEX: 55

Is this a hardware issue or is there something else I should try. It would appear the slave can see what's happening, but the Master can't see the slave??

Regards,
Graeme.

orgua, 2016-03-10:
hi @GraemeTownsend,
i tested the original atmega328 and a attiny85 so far. The Code is abstract enough to be hardware-independent. I took the platform-code from the onewire-lib and hoped it will work. 
i will look into this on the weekend. It is probably a timing issue. Maybe your oscillator is not configured right? Do you have a logic probe like a saleae logic 8? It would help to see the actual communication.
 
Just as a sidenote: the code is still a bit experimental - hope to reach v1 in the next week
Just for curiosity: are you using the latest code? i thought i took out the cmd-debug-output

GraemeTownsend, 2016-03-10:
Hi, Thanks for the reply.
I think I'm using the latest, but I will download again and start fresh, (I ran a lot of test's!).

Will have access to an oscilloscope Tuesday and will gather some good info for you. I will send it through then.
Thanks again.
Graeme.

GraemeTownsend, 2016-03-14:
Hi @orgua,

I've progressed some more. I downloaded the code again and started fresh.

Using "debug-master_DS18x20.ino" and "Sensor_DS18B20.ino". One change, in slave I set the LED pin to 9 to suit my board.

The two boards are talking, but not quiet right?

The slave output is consistently printing  out:
"Error: write timeslot timeout" - about once a second.

The Master is consistently printing out:
ROM = 28 D 1 8 B 2 A AB
  Chip = DS18B20
  Data = 1 FF FF FF FF FF FF FF FF FF  CRC=C9
  Temperature = -0.06 Celsius, 31.89 Fahrenheit
ROM = 28 D 1 8 B 2 B F5
  Chip = DS18B20
  Data = 1 FF FF FF FF FF FF FF FF FF  CRC=C9
  Temperature = -0.06 Celsius, 31.89 Fahrenheit
No more addresses.

The slave LED is blinking OK, but the temperature for SensorB is not changing.
I modified slave code:
"sensorA.setTemp(21);"
to 
"sensorA.setTemp(60);"
This did not change the  Master output from the above.

I'm going to start looking into the slave code now. Do you still want me to gather oscilloscope data for you?

Regards,
Graeme.

orgua, 2016-03-15:
Problem1: the serial-output of the master during the transmission is messing with the onewire-timing. 
Problem2: "retries"  in awaitTimeSlot() in OneWIreHub.cpp is most probably calculated wrong for your architecture. 
Fix: change "retries" to be always 60000 (Line 709 in the current code). 

I work on this problem. Using micros() for the timing is no option anymore, since we found out it takes about 3µs per call.

GraemeTownsend, 2016-03-15:
Hi,

Thanks.
I implemented the fix. Worked fine.

I even added some real DS18B20 to the bus and the Master just picked them up no problem.

I'm keen now to try the DS2408 and will let you know what I find.
Thanks again.

GraemeTownsend, 2016-04-26:
Hi,

I'm working on the DS2408 and have a quick query, I can't seem to send it a wrote command? Only the read command. Each time I send a write command an error is produced.

Looking at the DS2408.cpp file I can only see the read command in the switch statement. Am I missing something here?

Thanks,
Graeme.

orgua, 2016-04-26:
Yeah, it is not quite finished. The header file gives a hint in line 2. It doesn't look like much but i think the datasheet was a big pain. feel free to try :-)
i will hopefully continue to work on the lib by the end of the week.

GraemeTownsend, 2016-04-26:
Thanks. Yep, line 2 on header is quiet a hint... I missed it.
I've created the switch and it gets the hit, I just can't read the incoming data properly....

I'll check back at the end of the week.
Thanks.
(Let me know if you want me to do any testing)

orgua, 2016-04-26:
my todo-list has a lot of other stuff on it before i will have time for the ds2408. 
I'm curious - what's your test-setup? i wanted to compile a compatibility-list for successfully tested virtual sensors / devices.

GraemeTownsend, 2016-04-26:
Hi,

I'm running a duinotech LEONARDO as my Master.
My Slave is a Pro Micro, which is basically a cut down LEONARDO.

Both are set as Arduino LEONARDO's in the compiler.

orgua, 2016-11-07:
Can i close this issue?

GraemeTownsend, 2016-11-07:
Hi, 

 

Sure. Thanks.

 

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com> 
Version: 2016.0.7859 / Virus Database: 4664/13365 - Release Date: 11/07/16

orgua, 2016-11-07:
ok and just wanted to note: ds2408 is usable now.

closed 2016-11-07 by orgua