orgua/OneWireHub #3: ATtiny85

palsbo, 2016-03-20:
Hi - very nice work!

Has ATtiny85 been tested?

I have changed LED-pin to 3 and OW-pin to 2.
Using ATtiny85 8 MHz
Compile with Arduino IDE 1.6.7 - compile ok.
Upload using Arduino Nano. - upload ok

Led is blinking, but device is not recognized. on OW master (using Raspberry Pi with I2C for OneWire and OWFS).

I have another implementation that works. 
Details and code here http://palsbo.com/arduino/OneWireSlave.html

orgua, 2016-03-20:
You could try the latest release 0.9.0 from today. attinysupport is partly tested and should be improved now. had no chance to test my new code till now.

palsbo, 2016-03-20:
Hi - Thank you.
I have tried the new  version, and it is much better.

I do have som problems using OWFS. Some fields are not recognized ome 
times.  I suspect it is a timing issue, as it is not consistent.
Attached image show an example (r-locator).
Best regards
Erik

orgua, 2016-03-20:
image upload seems not to have worked

palsbo, 2016-03-21:
Sorry - try again:
[image]

palsbo, 2016-03-21:
I just tested with a DS9490 usb-stick. Here the ow-slave will not show at all.

palsbo, 2016-03-21:
After changing ONEWIRE_TIME_READ_STD in OneWireHub.h from 30 to 20, things seems to works much better on ATtiny85. Even with DS9490.  Not finished testing yet!

orgua, 2016-03-21:
Yes, this could help as a dirty fix and can even be reduced to 15 µs. switching between the functions() adds a large amount of additional overhead. 80 clock cycles make up 10 µs on your platform. i will reduce this overhead-footprint in future commits. 
We can't just switch to ASM, because it wouldn't be platform-independent anymore. 
For the next 3-4 weeks i won't have hardware to test new code, so there will be no commits from my side. Feel free to research this topic in the meantime.
I collected more timing-related information from various datasheets in an older version of the onewirehub.h : 
https://github.com/orgua/OneWireHub/blob/3e67d91e2ee7c040fbeb315949272b38393196c5/src/OneWireHub.h

orgua, 2016-09-05:
is there something new on your side? i am beginning to continue my work on the lib

orgua, 2016-11-07:
closing because thread is inactive.

closed 2016-11-07 by orgua

palsbo, 2017-12-06:
Sorry for the big delay - there was a reason!
In the meantime there are new version and I cannot get it to work with ATTiny85 (and DS9490)
I tried to look into the timing, but nothing helped.
Any suggestions?