orgua/OneWireHub #6: DS2501 and DS2502

xcvista, 2016-06-25:
Those chips are DS2433 with an extra CRC instruction. Maybe they can be implemented based on DS2433 code?

Those devices are used in Dell laptop power supplies however they fry too often. This is where emulating them comes in handy.

TylerDurden23, 2016-08-16:
https://forum.pjrc.com/threads/33640-Teensy-2-OneWire-Slave

orgua, 2016-08-27:
so i conclude you don't have any coding-experience? i will set it on my todo - if it is doable in <2h i will try

xcvista, 2016-08-31:
@orgua I went into your DS2433 code and had some trouble parsing the logic in there.

orgua, 2016-09-05:
ok, i looked at it and they seem to be similar. i will put the ds2433, ds2501/02 on my list. the hack in the forum is now included in the lib, but its only a subset of the ds2502 but it seems to work with the dell

orgua, 2016-09-18:
ok, done

closed 2016-09-18 by orgua

sanjeevirm, 2023-03-04:
We have procured DS2502 and DS28E07+ and we have facing the writing programming issue.

We have connected Arduino uno to 1 wire DS2502 or DS28E07+ device but we can't write the device.

Please help us how the set up we need make and how to write the 1 Wire EEPROM. or (please share me the setup image)

For actually our main aim is to write the dell charger ID value into the EEPROM (Charging the laptop)

Please help us we have struggled last one month . please provide the email ID

orgua, 2023-03-07:
This lib simulates these devices and does not talk to them. 

- [here](https://github.com/orgua/OneWireHub/tree/main/examples/DS2502_DELLCHG) is working code to emulate a dell charger with a virtual DS2502.

There are dedicated onewire-host-libs for your usecase (if I understand you correctly):

- [my take to talk to the DS250x EPROMS](https://github.com/orgua/OneWire/blob/master/examples/DS250x_PROM/DS250x_PROM.ino) using the onewire-lib

- more recommended is the modernized [onewireNG](https://github.com/pstolarz/OneWireNg) for talking to sensors, but you have to write the code yourself