orgua/OneWireHub #7: DS2431 support

j-langlois, 2016-08-26:
dear friend,
thanks for the great work! definitely the best codebase I've seen to simulate 1-wire slaves. I've been playing around with the library for the last few days, and have successfully implemented all the operations for the DS2431 1kb EEPROM, tested on mega 328p. However, I have found out that for a particular master, the slave will have timing issues, thowing WRITE_TIMESLOT_TIMEOUT errors. I've been playing around adjusting the delays from the manufacturer's datasheet without any success. Any hint would be greatly appreciated. I intend on cleaning my code and submitting a pull request later on.  
Thanks
Joel

orgua, 2016-08-27:
it would help if you could give some more information: 
- is the at328 running with 16 MHz?
- which arduino-version do you use? the latest >1.6.10 is still untested by me but compiled code should be superior to prev. versions
- do you have timing-issues when debug-output is disabled?
- could you include a link to your current code? maybe i can find a clue. maybe do a pull request with a "do not merge"-mark.

j-langlois, 2016-08-27:
thanks for the prompt response
I'm at 16Mhz on arduino 1.5 and yes the outputs are off. I might have isolated the issue yesterday. The problem seems to be related to the reset detection surrounding some arbitrary number of read operations the master can execute following the READ MEMORY 0xF0 command. By that, I mean that it could do read X bytes, reset the bus and read Y bytes afterwards. My working trials did not include sequential arbitrary read operations separated by reset commands. I will be out for a few days but I will come back to you on this with some code.
Cheers

orgua, 2016-08-27:
now i am wondering how you could compile this lib. it uses c++-features that are not supported by default in arduino <1.6.x. you have to enable them manually in a config-file. please try updating it. 
by using 1.6.10 instead of 1.6.8 i got around 10% smaller codesize. this could mean that timing constrains are a bit more relaxed now. 

the behaviour you have seen should be fixed in my last commits thou :-( ("hub is more robust to odd master-behaviour (lazy timings and subsequent resets)"). I've seen them too with the ds9490.

j-langlois, 2016-09-01:
Thanks for the tip! Indeed I used 1.6.10 and got it to work flawlessly!

orgua, 2016-09-01:
nice to hear! hope to see your pullrequest soon :-) 
An included example would be much appreciated! 
I would offer my help if you need help with the cleanup.

j-langlois, 2016-09-02:
Sure you can close it! PR will be in in an hour or so
Cheers

orgua, 2016-09-03:
i leave this open to remind me of testing it. you mentioned another OW-master. have you tested other sensors of this libs? what kind of system is it?

j-langlois, 2016-09-03:
The master I was referring to is a PIC18. It is part of an industrial machine using the DS2431 as storage, so many R/W operations involved.

closed 2016-09-05 by orgua