orgua/OneWireHub #10: DS2433 slave not visible to master on parasite power

mjbraun, 2016-09-18:
I'm running an Arduino Mega ADK as a DS2433 and when the I wire up the Arduino with a 4.7k resistor and connect it to a Maxim BlueDot reader attached to a DS9490R USB port adapter, everything works fine. However, when I attach leads to the unidentified-model reader built into a third-party linux device that I'm trying to interact with, the DS2443 is not shown. 

I'm seeing READ_TIMESLOT_TIMEOUT errors when I enable serial debugging and using a logic analyzer to analyze 1Wire protocol  I see the reset signal as being slightly too short (~475usec vs 480; not sure if that matters). If I have the BlueDot master wired in as well, the BlueDot can interact with the device but the unidentified-model reader cannot.

I suspect the issue is related to the linux device being in parasitic power mode. A DS1822 thermometer IC elsewhere on the bus registers on the device, so I know the bus is functional. Are there special steps required to enable compatibility with parasitic power mode or are there parameters that should focus on adjusting? Thank you!

orgua, 2016-09-18:
well, i try to set two things straight: 
- there can always only be one master on the bus (otherwise you get undefined behaviour)
- i doubt that you can use your Atmega without additional powersource
   - you should at least put it to sleep between polls(), but then it could miss resets 
   - your input-capacitor for your ATMEGA should at least buffer 1ms with ~10mA current drawn

If you are powering the atmega with an USB-BUB (you mention reading serial) then it should work if you have set the CPU-Frequency correctly and the two devices are connected by GND- and Signal-Wire

mjbraun, 2016-09-19:
Thanks for responding! (Also, I typo'd: it's a DS2431) 

I was only running the dual master setup to simplify switching back and forth from the "known-good" BlueDot master to the misbehaving linux device. Running the linux device as the only master is normally how I'd operate.

The Mega is powered by a 12V 1.5A power supply. GND and signal are connected directly (I also tried with the inline resistor), but I'm not sure which CPU-Frequency parameter you're referring to: is this something in platform.h?

orgua, 2016-09-19:
without a logic analyzer it is hard to say whats wrong. With CPU-Freq i meant, that on some boards you can choose the baseclock. but yours is 16 MHz. it's fine! And the ADK should be supported, but is untested.
Please try the following:
- make sure your arduino-software is up to date (>v1.6.8)
- update this lib to the latest release (0.9.7)
- disable UART / GPIO debug in onewirehub.h (uart can cause missing timeslots)
- power the atmega-board per usb or any other safe source
- load the ds2401 example (with matching pins) to the atmega
- connect your atmega directly to the ds9490R (signal + gnd) 
--> if this works continue with the ds2431 (i updated the code quite a lot over the weekend)

this is working for me with a standard arduino (atmega328p)

mjbraun, 2016-09-19:
I updated the library in the Arduino software (1.6.11) to the 0.9.7 release, USE_SERIAL_DEBUG and USE_GPIO_DEBUG are #defined to 0 in ~/Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h. USB and 12V power provided to the Arduino. Connections are wired directly (no resistor or 5V from the Arduino).

A physical iButton registers on the Linux device but the Arduino loaded with the ds2401 example and wired from GND and pin 8 do not register. I have a Saleae logic analyzer tapped into the connection and here are the two captures from the iButton and the Arduino-as-serial (note there is also a DS1822 on the bus). I think the master is misbehaving, as in the Saleae Logic application, the 1wire protocol is decoded and shows the RESET pulse as being too short. 

Note that there is no expected FAMILY CODE response to the SEARCH_ROM command for the Arduino whereas with the iButton the FAMILY_CODE response is detected.

[Captures.zip](https://github.com/orgua/OneWireHub/files/481177/Captures.zip)

Thank you so much for your advice/guidance!

orgua, 2016-09-19:
the linux device is really edgy. 
The communication after 1.823435s in the second capture could be your atmega. but could also be anything else. now that i know that you can debug, you should use the gpio-debug and add some markers in the code. in the current state you would get a pin-high during the call of slave.duty(). you could also do that for showPresence() in the poll()-FN.

mjbraun, 2016-09-23:
I'm a bit out of my depth here, so I appreciate your patience. I noticed in recent captures the linux device holds a reset condition for 1.194ms, so I upped ONEWIRE_TIME_RESET_MAX and got things moving a bit but now I'm not decoding cmds (they're only decoded as "0"0. Tracing down I see errors in awaitTimeSlotAndWrite (Error: read timeout high) which seems to happen because DIRECT_READ never returns 0. Have you seen this sort of error before?

closed 2016-10-18 by mjbraun