orgua/OneWireHub #9: crc16 of the ds2431

orgua, 2016-09-18:
hi @j-langlois 

could you give me hint where you got the start-value of the crc. the datasheet is missing some information. and does not mention anything else than flush/reset the crc16-input. i have extended the device and everythings works, except the eprom-mode. 
i had trouble using your start-value and more luck with just 0.
ds2433 is very similar but doesn't mention any information either.

j-langlois, 2016-09-18:
Hello orgua,
Honestly, I do not have a clue why no start value is mentioned in the documentation. Basically, I made several writes to the scratchpad of an actual DS2431 and noted what the chip gave me. Then, I ran every start value in the code to determine which one would match the result given by the chip. However, this might very well be version specific. I am using DS2431Gs.

What issues are you experiencing with the eeprom mode and master are you using ?

orgua, 2016-09-18:
oh well, so this could be a crc-collision. did you begin with 0x0000 for your brute-force-calculation? It should work. the DS9490 reads it everytime.
using the ds2431-slave with the current code and a ds9490R as a master i get: 
- changing the pages and upper control-bytes work as expected when no matching write protection is set
- using/changing pages in EEPROM-MODE gives me a bad CRC and no data seems to be written/cleared

orgua, 2016-11-07:
ok, EPROMs seem to initialize the CRC with zero and not with some magic-byte. my master ds9490 is ok with that. will close this

closed 2016-11-07 by orgua