orgua/OneWireHub #19: DS2423 support

trycoon, 2016-10-25:
Hi, as far as I can see there are some initial support for DS2423 in the sourcecode. But there are currently no way to set the counters to a value. https://github.com/orgua/OneWireHub/blob/master/src/DS2423.h

As a newbee OneWire user and C++ developer it's hard for me to tell what have been done and what's left. No memory seems to be reserved for the 16 pages of 256 bit memory that the device should have.

I'm looking for the simple external counter support of the DS2423, does anyone out there have time to do a first implementation then I would gladly test it for you. Thank!

orgua, 2016-10-25:
it is hard to do something for free. maybe you can use code of the ds2433 by comparing the datasheets. it offers 4kb EEPROM. the original code is here - it should be the same, but outdated: https://github.com/orgua/OneWireHub/blob/90c3a9acc6ea174739c4510f5376e75c1ae5ce3a/src/DS2423.cpp 
As i see it is only a mockup so far. All read data equals 0xFF, counter is always 0x00. But it works with a ds9490-master. just tested

orgua, 2016-10-25:
maybe as a starting point - here is the example-sketch i just made: https://github.com/orgua/OneWireHub/blob/tick_timing_and_overdrive/examples/DS2423_RAM/DS2423_RAM.ino

orgua, 2016-10-26:
i looked at the datasheet and the ds2433 is indeed very similar. after finishing the overdrive-support i could look into this. is it very urgent?

trycoon, 2016-10-26:
Den 2016-10-25 kl. 23:39, skrev inʒo:

> it is hard to do something for free. maybe you can use code of the
> ds2433 by comparing the datasheets. it offers 4kb EEPROM. the original
> code is here - it should be the same, but outdated: [...]
Yes I understand that, the DS2433-code looks more complete. That would 
be a better starting point.

Thanks!

trycoon, 2016-10-26:
Thanks, I'll see if I can fill it with more example code.

// Henrik

Den 2016-10-25 kl. 23:42, skrev inʒo:

trycoon, 2016-10-26:
No I wouldn't say that its urgent, just for a hobby project of mine. I could use a DS2413 to signal that an external event has happened, but a counter is more convenient. Using a DS2413 I have to poll it alot to not miss an event, using a DS2423 counter I can read the counter whenever I want, if I know that the value was 36 the last time I checked and its now 39 I know that three events has occured since the last time (unless the circut has been powercycled).

Too bad Maxim has stopped producing these gems, I hope they will make a replacement part in the future.

Yes, It's much like the DS2433, but it has four counters, so some readCounter(counter) and setCounter(counter, value) methods are needed.

orgua, 2016-10-26:
i am glad you mentioned your application. the counters in the ds2423 are used to count the write-operations to the memory-banks. it's possible to verify the data inside it and make sure it wasn't altered. you don't need that. 
But speaking of emulation: you can put any data in these slaves. they don't care. example: 
- use four emulated ds18b20 and you can use the 12bit temp-value as a counter. just write your value with setTempRaw() into each slave when you want to count something (you will have to make this function public instead of private to use it in the ardu-sketch
- use one ds2431 or ds2433 and write your counter-values with writeMemory() to certain positions. if you want to be safe you can disable the "write scratchpad command" and your master won't be able to alter your data

Does this help?

trycoon, 2016-10-27:
The some of the counters in DS2423 also react to external events like 
pulses on a pin to the DS2423.

Hahaha, nice abuse of the hardware using the DS1820 as a counter, but 
that would solve my usecase. Many thanks for your help!

// Henrik

Den 2016-10-26 kl. 19:00, skrev inʒo:

trycoon, 2016-10-30:
Hi again!

I have soldered up a simple board with an ATmega 328P, a 433MHz receiver 
and a LED-light. The ATmega emulates a DS2433 and counts every signal it 
receives from the 433MHz receiver and save the counter value to the 
DS2433 memory and flashes the LED. The board is then connected to my 
OneWire network at home. All the receiving and LED-flashing does work, 
but scanning my OneWire network with "owdir /uncached" does not reveal 
the emulated DS2433. Have you any suggestion about what could be wrong? 
I have followed the example within the OnewireHub project, but maybe I'm 
missing something? The "scan ROM"-command is supported within 
OneWireHub, right?

https://github.com/trycoon/mailbox_reader/blob/master/mailbox_reader.ino

Thanks!

// Henrik

orgua, 2016-10-30:
try emulating a simple sensor like the ds18b20. it doesn't support overdrive, so your master won't try switching to higher data rates. it should at least pop up. if not -> your setup is wrong. Is the data-line really on pin D8. Is ground connected? Timing is critical, so your boards clock should match the setting in your software. You could test this by blinking a led with a 1sec pause. it is easy to see if it takes 2s or just 0.5s on your arduino

trycoon, 2016-10-30:
Thanks, I'll try that later. Also I'm not running in parasitic mode, I use all three pins, if that matters. How do I set the boards clock? I use a normal 16Mhz crystal, is there a software setting for that?

//Henrik

trycoon, 2016-10-30:
Den 2016-10-30 kl. 10:09, skrev inʒo:

> try emulating a simple sensor like the ds18b20. it doesn't support
> overdrive, so your master won't try switching to higher data rates. it
> should at least pop up. if not -> your setup is wrong. Is [...]
Ok, I got some progress. I tried to change to a DS18S20 but still could 
not find the emulated sensor. I then commented out all but the most 
essential code and then my computer found the DS18S20 with a temperature 
of 33 degress. Hurrah! I then switched back to a DS2433 with some 
hardcoded data in memory and it worked, Hurrah again! So my timings and 
cabling seems Ok, what a relief.

Soo I guess the library that I use to recieve data from a 
radiotransmitter may be the one causing trouble, or some of my other 
still commented lines of code.

How often to we need to call upon the poll()-method? If any of my code 
or the libraries that I use block for a long time, could that not 
prevent the OneWireHub from working? Or does OneWireHub use interrupts 
to handle the critical stuff?

Thanks!

// Henrik

trycoon, 2016-10-30:
Den 2016-10-30 kl. 10:09, skrev inʒo:

> try emulating a simple sensor like the ds18b20. it doesn't support
> overdrive, so your master won't try switching to higher data rates. it
> should at least pop up. if not -> your setup is wrong. Is [...]
Reading the memory multiple times gives you different results every 
time, is this somekind of timing issue or is this correct behavior?

Since I run "ds2433.clearMemory();" at start I expected it to be all zero.

owread -s 192.168.10.110 /uxdump -C23.DEADBEEF1010/memory |hexdump -C
00000000  00 02 00 00 00 00 00 00  00 02 00 02 00 00 00 00 
|................|
00000010  00 00 02 00 02 00 00 00  00 00 00 00 00 00 02 00 
|................|
00000020  02 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000030  00 00 00 00 00 02 02 00  00 00 00 00 00 00 00 00 
|................|
00000040  00 00 00 00 00 00 00 00  00 00 02 00 00 00 00 00 
|................|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000060  02 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000080  00 00 00 00 00 00 00 00  00 00 00 00 00 02 00 00 
|................|
00000090  00 00 00 00 00 00 00 00  02 00 00 00 02 00 00 00 
|................|
000000a0  00 00 00 00 00 00 00 02  00 00 02 00 00 00 00 00 
|................|
000000b0  00 00 00 00 00 00 00 00  00 00 00 00 00 02 00 00 
|................|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
000000d0  00 02 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
000000e0  00 00 00 00 00 00 00 00  00 00 02 00 00 00 00 00 
|................|
000000f0  00 02 00 00 00 02 00 00  00 00 00 00 00 00 00 00 
|................|
00000100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000110  00 00 00 00 00 00 00 00  20 00 00 00 00 00 00 00 |........ 
.......|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
*
00000140  00 02 00 00 00 00 00 00  00 00 00 00 00 00 02 00 
|................|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000160  00 00 00 00 02 00 00 00  00 00 00 02 00 00 00 00 
|................|
00000170  00 00 00 00 00 02 02 00  00 02 00 00 00 00 00 00 
|................|
00000180  00 02 00 00 00 00 00 00  00 00 00 02 00 00 00 00 
|................|
00000190  00 08 10 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
000001a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
000001b0  00 00 00 00 00 00 00 00  02 02 00 00 00 00 00 00 
|................|
000001c0  00 00 00 02 00 02 00 00  00 00 00 00 00 00 02 02 
|................|
000001d0  02 00 00 00 00 00 00 00  00 00 00 00 00 00 00 10 
|................|
[... 100 lines cut ...]
00000130  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
*
00000160  00 00 00 00 00 02 00 02  00 00 00 00 00 00 00 00 
|................|
00000170  00 00 00 00 00 00 40 00  00 00 00 00 00 00 00 00 
|......@.........|
00000180  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
00000190  00 00 00 00 02 00 00 00  00 00 00 00 00 00 00 00 
|................|
000001a0  00 00 00 00 00 00 00 00  00 00 00 02 00 00 00 00 
|................|
000001b0  00 00 00 00 00 00 02 00  00 00 00 00 00 00 00 00 
|................|
000001c0  00 02 00 00 00 00 00 00  00 00 00 00 02 00 00 00 
|................|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 
|................|
*
000001f0  00 00 00 00 00 02 00 00  00 00 00 00 00 00 00 00 
|................|
00000200

// Henrik

orgua, 2016-10-30:
ok, some additions to your problem: 
- memory should be zero, but what you see there are sporadic bit-flips. your µC seems to be to fast or your master is edgy
   - try to increase  `static constexpr uint16_t ONEWIRE_TIME_WRITE_ZERO_LOW_STD = 30;` to about ~35 or slightly more
- yes, i meant try a basic example first - without your custom code. 
- yes, polling() is bad and two devices polling() is even worse. no irq-handling here at the moment. so you have to look for a proper replacement for your rf-lib or write some kind of scheduler. bad

trycoon, 2016-10-31:
>       o try to increase |static constexpr uint16_t

I you this for master: https://www.axiris.eu/en/index.php/1-wire/abiowire
I tried to increase to 35, but could not see that much difference, at 40 
the sensor was not found at all. I decreased to 25 but got about as much 
static as at 35.
I will leave it at 25 for the moment, by reading /pages/page.0 several 
times in a row I could filter out the bit-flips I will get a steady reading.

I belive the rf-lib works purely with interrupts (int #0) now that I 
look at its code, soo OneWireHub is the only one polling for now.

Many thanks for all help so far!

// Henrik

Den 2016-10-30 kl. 18:44, skrev inʒo:

orgua, 2016-11-07:
did you have any luck finding a solution to your problem? btw, the newest branch contains the ds2423: 
https://github.com/orgua/OneWireHub/tree/tick_timing_and_overdrive

trycoon, 2016-11-08:
I haven't changed my code since the last posting, but it works quite nicely now. The implementation looks feature complete in the timing-branch so I probably switch to the DS2423-device once a new release gets out. You can close this issue if you want. Many Thanks!

orgua, 2016-11-10:
feel free to reopen if there are any troubles migrating to the new v2

closed 2016-11-10 by orgua