orgua/OneWireHub #20: OneWire device with family code B3

256FF, 2016-11-02:
I'm want to emulate a onewire device of my printer chip. It is alike the DS2433 but chip family is B3. 
Can anyone help me? In the attachment there are some dumps of the EEPROM.

[ori_1.bin.txt](https://github.com/orgua/OneWireHub/files/566180/ori_1.bin.txt)
[ori_2.bin.txt](https://github.com/orgua/OneWireHub/files/566179/ori_2.bin.txt)

orgua, 2016-11-02:
You can read out the eeprom but can't solve this task? Seems odd. Just load your *.txt into notepad++ with hex-editor-plugin or any other similar hex-editor and copy your data into an uint8_t array and write it into your emulated device with the default DS2433-Example:

`
auto hub = OneWireHub(pin_onewire);
auto ds2433 = DS2433(0xB3, 0x01, 0x01, 0x33, 0x24, 0xD0, 0x00);

constexpr uint8_t memSTRATASYS[] = {
        0xed, 0x45, 0x85, 0x80, 0x6b, 0x9b, 0xa1, 0x1e, 0xf3, 0x20, 0xbb, 0x2e, 0x2f, 0x96, 0x74, 0xa9,
        0x67, 0xf2, 0xb9, 0x2a, 0x56, 0xc7, 0x79, 0xf6, 0x90, 0xc2, 0x42, 0x7f, 0xcb, 0x09, 0x60, 0x53,
        0xf2, 0x05, 0x72, 0x63, 0x08, 0xf0, 0x36, 0x72, 0xec, 0xf2, 0x5d, 0x53, 0x52, 0xe3, 0x6d, 0x74,
        0xec, 0xf2, 0x5d, 0x53, 0x52, 0xe3, 0x6d, 0x74, 0xb0, 0x89, 0x5c, 0x51, 0x69, 0xae, 0x14, 0xc5,
        0x22, 0x60, 0x20, 0x20, 0x20, 0x20, 0x74, 0x25, 0x55, 0xaa, 0x55, 0x83, 0x8b, 0xf7, 0x4c, 0xbe,
        0xa8, 0x74, 0x20, 0x20, 0x20, 0x20, 0x20, 0x20, 0x7e, 0x35, 0x04, 0x51, 0x0f, 0x0e, 0x1c, 0x9e,
        0xf4, 0x5c, 0x56, 0xf3, 0xfc, 0xa9, 0xf1, 0xd2, 0x53, 0x54, 0x52, 0x41, 0x54, 0x41, 0x53, 0x59,
        0x53, 0xac, 0x99, 0xcb, 0xee, 0x48, 0xb4, 0x9e, 0xa3, 0x62, 0xf0, 0x97, 0xa5, 0x2b, 0x84, 0x38,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x20, 0x55, 0x20, 0x55, 0x20, 0x20, 0x20, 0x20,
        0xb3, 0x5c, 0x1c, 0x0b, 0xd0, 0x14, 0x10, 0x0d, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
        0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff };

void setup()
{
    Serial.begin(115200);
    Serial.println("OneWire-Hub DS2433");

    // Setup OneWire
    hub.attach(ds2433);

    ds2433.writeMemory(memSTRATASYS, sizeof(memSTRATASYS), 0);

    Serial.println("config done");
}

`

I just had to replace the spaces " " with ", 0x" after copying the hex-values....
Code is untested but should work. The alternative is to directly modify the memory by changing the constructor in the ds2433.cpp / replace clearMemory()....

256FF, 2016-11-02:
Thanks for the description. Did not realize that it is so easy to add the content.

orgua, 2016-11-02:
I didn't want to be offensive, but it looked like you solved 90% on your own. And i think the previous task was harder.

256FF, 2016-11-02:
No problem. ATM I'm a bit familiar with the Stratasys Chips but reading up on to get knowlegde with the arduino. I'll be back with more questions soon. ;-P

256FF, 2016-11-04:
Back again. 
Now I'm juggling with the UID. When dumping the EEPROM I get 8 Double Hex blocks of the original Chip. 
`b3 bb dc 06 d0 14 10 1e`

But in the DS2433.h I only have 7 blocks available. Do I have to exchange https://github.com/orgua/OneWireHub/blob/master/src/DS2433.h#L15 to family_codd = b3 ? and then add the other 7 blocks to the other file? 

This means instead of
`auto ds2433 = DS2433(0xB3, 0x01, 0x01, 0x33, 0x24, 0xD0, 0x00);`

the line should look like this:
`auto ds2433 = DS2433(0x01, 0x01, 0x33, 0x24, 0xD0, 0x00, 0x11);`

orgua, 2016-11-05:
I think i will close this issue for now since there are no problems regarding the lib itself. Your problems are more suited for an arduino-forum.
Answers for your c/c++-questions should have been already discussed at stackoverflow. For the questions regarding onewire even wikipedia would suffice. And nearly every slave-datasheet answers the ROM / UID Question on the first 3 pages.

closed 2016-11-05 by orgua