orgua/OneWireHub #23: CRC problems with SparkFun Pro Micro

hJcRox, 2016-12-04:
Hi
I am trying to use a SparkFun Pro Micro @16 MHz with the OneWireHub 2.0.
I am using the DS18B20 example with only one sensor to test. The problem I have is CRC8 errors when reading temperature, about 5-15% failure rate.
I have tried various things to try to understand the problem and its cause:

- I have made a calibration giving VALUE_IPL=13. Result text as below:
DEBUG TIMINGS for the HUB (measured in loops):
(be sure to update VALUE_IPL in src/OneWireHub_config.h first!)
value : 	875 nanoseconds per loop
reset min : 	491
reset max : 	1097
reset tout : 	5714
presence min : 	22
presence low : 	182
pres low max : 	548
msg hi timeout : 	17142
slot max : 	154
read1low : 	68
read std : 	22
write zero : 	34

- Checked other VALUE_IPL as well, <12 not working at all. 12 and 13 about the same, and 14 is worse, but still working.
- At CRC error it seems like my master (Windows with DS9490R) is reading all 0xff.
- Most testing done with out serial debug but when on I get "Error: first bit of byte timeout" when CRC exemption
- Tested a program a pure loop using micros() for the delay. With 9995 us between toggle a readout 50.0 Hz with my multi meter (not sure about its accuracy). 
- I have no oscilloscope and can not check the 1-wire waveforms.
- Using 1.6.11 of Arduino.
- I have tested with a real DS18B20 and it worked perfekt. No errors when running ~5000 temperature readings

Would be fine with some hints on how to continue debugging. The possibilities with this SW seems great. But it must be more reliable.

Regards Jan

hJcRox, 2017-01-02:
Hi again
I found the problem. It seems to be a combination of a too short time-out ONEWIRE_TIME_MSG_HIGH_TIMEOUT = 15000_us
and the Dallas Java 1-wire API sometimes running too slow. This caused CRC8 errors and "Error: first bit of byte timeout" traces in the Arduino monitor.
After changing to ONEWIRE_TIME_MSG_HIGH_TIMEOUT = 150000_us everything seems to be perfect. No CRC8 errors and no error-traces in the monitor.
Regards Jan

orgua, 2017-01-31:
this behavior seems very odd but i have never worked with a ATmega32U4. its hard to say anything without  a logic-analyzer on your side. sometimes you get these logic-8 very cheap. but besides that you have done everything you could do! Very good. 
Changing this one value should be fine, but i have the same ds2490 and it never needed a higher timing. it would only hurt when your µC has to do other stuff. polling is very slow this way.

orgua, 2017-08-20:
ok, made a remark in readme in section "### HELP - What to do if things don't work as expected?"

closed 2017-08-20 by orgua