orgua/OneWireHub #25: Raspberry Pi compatbility

chasm-git, 2017-02-05:
Using GPIO4 that is.
Has someone managed to read emulated senors with Raspberry Pi via GPIO?

Hardware 18B20 on either side of a level shifter work as expected, the emulated sensors are a no-go. (Arduino nano, Arduino pro mirco 3.3V, ESP8266)
Reading the emulated sensors via various software bus masters works. 

Switch to DS2482 / DS2484 as physical bus master. - Or find the mistake?

orgua, 2017-02-07:
did you translate your text via google translate? it is hard to understand without guessing a lot.

so you seem to have a RPI as a onewire-master and want to emulate slave via arduinos. it should work. It depends on the software of the RPI. they all try to speak onewire and when someone is bending the rules we all have a problem. 
you offer little to no information for me to help you. i am sorry. 
and yes, there was someone who used two rasperries to speak to each other, one as a master, the other had the onewirehub installed. it went smoothly, but surely needed some knowledge on debugging

chasm-git, 2017-02-12:
Ooops. A bit too much last minute editing and not enough proofreading.

I use a Raspberry Pi to read a string of DS18B20 temperature sensors and log the data into a DB. This has been running for years. First polling the sensors over the network with an ethersex board (think Arduino with ethernet shield), then directly attached to GPIO4.

Now I want to add humidity sensors into the mix. The simplest way should be to emulate them as 1-wire temperature sensors. The existing wiring is able to power an Arduino.

Small steps first. I put the DS18B20 example on an Arduino nano and tried to read it with another Arduino. This works as expected.

Then I tried to read the emulated sensor with the Raspberry Pi. - This does not work. I can still read the physical DS18B20 sensors on the same bus. No obvious errors. Supply voltage is ok.

A few hours later I come to the conclusion that I can read OneWireHub sensors reliably with various micro controllers over different distances but never with the Raspberry Pi.
That seemed to be a good point to stop and ask if someone else has done it before.

Next week I can remove GPIO4 from the equation, going for DS2482-100 or DS2483 instead.
One of the cheap logic analysers is also on order. Hopefully this one will arrive. 3rd time is the charm...

chasm-git, 2017-02-17:
Switched over to a DS2482-100 in and it works. So the problem seems to be GPIO4 specific.

closed 2017-08-20 by orgua

sashad, 2019-03-19:
Hello, I want to use DS2482-100 on esp8266 (esp12) with OneWireHub library. Is it posible?