orgua/OneWireHub #28: issues with Wemos D1 Mini

toabhishekverma, 2017-03-01:
Hey Guys,
I was trying to use this library with Wemos W1 D1 Mini (Basically its ESP8266)

But I got Tons of error. I am new to Arduino and cant understand the error. 
Can anyone help me to out to find they problem. Here is the error s 

Arduino: 1.8.1 (Linux), Board: "WeMos D1 R2 & mini, 80 MHz, 115200, 4M (3M SPIFFS)"

/home/uptopoint/Software/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/uptopoint/Software/arduino-1.8.1/hardware -tools /home/uptopoint/Software/arduino-1.8.1/tools-builder -tools /home/uptopoint/Software/arduino-1.8.1/hardware/tools/avr -built-in-libraries /home/uptopoint/Software/arduino-1.8.1/libraries -libraries /home/uptopoint/Arduino/libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10801 -build-path /tmp/arduino_build_13235 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /home/uptopoint/Workspace/OneWireHub/examples/DS2431_EEPROM/DS2431_EEPROM.ino
/home/uptopoint/Software/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/uptopoint/Software/arduino-1.8.1/hardware -tools /home/uptopoint/Software/arduino-1.8.1/tools-builder -tools /home/uptopoint/Software/arduino-1.8.1/hardware/tools/avr -built-in-libraries /home/uptopoint/Software/arduino-1.8.1/libraries -libraries /home/uptopoint/Arduino/libraries -fqbn=esp8266com:esp8266:d1_mini:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10801 -build-path /tmp/arduino_build_13235 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /home/uptopoint/Workspace/OneWireHub/examples/DS2431_EEPROM/DS2431_EEPROM.ino
Using board 'd1_mini' from platform in folder: /home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266
Using core 'esp8266' from platform in folder: /home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266
WARNING: Spurious .settings folder in 'OneWireHub' library
Detecting libraries used...
"/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_13235/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/variants/d1_mini" "/tmp/arduino_build_13235/sketch/DS2431_EEPROM.ino.cpp" -o "/dev/null"
"/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_13235/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/variants/d1_mini" "-I/home/uptopoint/Arduino/libraries/OneWireHub/src" "/tmp/arduino_build_13235/sketch/DS2431_EEPROM.ino.cpp" -o "/dev/null"
"/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_13235/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/variants/d1_mini" "-I/home/uptopoint/Arduino/libraries/OneWireHub/src" "/home/uptopoint/Arduino/libraries/OneWireHub/src/BAE910.cpp" -o "/dev/null"
"/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_13235/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/variants/d1_mini" "-I/home/uptopoint/Arduino/libraries/OneWireHub/src" "/home/uptopoint/Arduino/libraries/OneWireHub/src/DS18B20.cpp" -o "/dev/null"
"/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_13235/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/variants/d1_mini" "-I/home/uptopoint/Arduino/libraries/OneWireHub/src" "/home/uptopoint/Arduino/libraries/OneWireHub/src/DS2401.cpp" -o "/dev/null"
"/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/lwip/include" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/arduino_build_13235/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10801 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"  -DESP8266 "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/cores/esp8266" "-I/home/uptopoint/Software/arduino-1.8.1/hardware/esp8266com/esp8266/variants/d1_mini" "-I/home/uptopoint/Arduino/libraries/OneWireHub/src" "/home/uptopoint/Arduino/libraries/OneWireHub/src/ [... cut ...]
/home/uptopoint/Arduino/libraries/OneWireHub/src/OneWireHub_config.h:39:67: error: 'constexpr uint32_t operator"" _us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_WRITE_ZERO[2]        = {  30_us,  8_us }; // the hub holds a zero for this long
                                                                   ^
Using library OneWireHub at version 2.0.0 in folder: /home/uptopoint/Arduino/libraries/OneWireHub 
exit status 1
Error compiling for board WeMos D1 R2 & mini.

orgua, 2017-03-02:
it looks like a known problem. there are two possible ways for you - try to understand and implement the fixes given in #22 or simply wait until i have some spare time to release a new version with tested code. the esp and its underlying OS is not well suited for timing-applications like this onewirebus

orgua, 2017-08-20:
tested hub with a wemos d1 r2, nodeMCU 1.0 ESP-12E and even a esp32
they all work!

closed 2017-08-20 by orgua