orgua/OneWireHub #29: ATtiny85 one wire issue

dakarym, 2017-03-12:
I want to emulate a DS18B20 sensor from an ATTiny85.  I have followed your instructions the best I can but fail to get it to work.  Using an Uno, I can try to read the OW temperatures.  The device number (ROM) is ok, but the temperature data is erratic and coming through as mostly FF's.  

1) Latest master repository from git
2) 1.8.1 version of Arduino IDE
3) Use serial debug and use gpio debug both set to 0 in onewirehub_config
4) A real DA18B20 sensor works properly with my OW master Uno
5) Your "calibrate_by_bus_timing" sketch does not seem to run on either my ATtiny85 or Uno, unless I am not using it correctly.  It just shows report the "OneWire-Hub calibration by observing the OW-Bus" via serial.  nothing more, and the led pin doesn't blink
6) The attiny85-4slaves.ino debug example fails to compile due to an ambiguious error
7) I can see that the ds18b20 sensor is receiving and reporting the correct number via the below lines in the test sketch, but the digital value via OW is not correct.

    mySerial.print("Test - set Temperatures to 0 degC: ");
    ds18b20.setTemperature(int8_t(0));
    mySerial.print(ds18b20.getTemperature());

I have a scope and can look at the one wire signal, but I do not know what feature to monitor or which of the timing values to change.

dakarym, 2017-03-12:
I put a scope on the bus.  Here is the OW bus messages, this is just repeating
[image]

Message detail of 'shorter' message
[image]

Message detail of 'longer' message
[image]

orgua, 2017-08-20:
oh man, i missed your issue. sorry. were you able to solve your problem? 

i can give you some hints for your problems:
- calibrate by bus-timing has the following setup: upload sketch to controller and hook it up to a OW-master. controller will listen to the bus and determine value_ipl by measuring the resetpulse
   - but this won't work with your attiny, because it should have no serial
- you are right, the "attiny85-4slaves.ino" example was broken due to changes in data-types, newest lib-version works now

with which attiny-library are you working? arduino itself does not support these chips.
did the compiler tell you, that your value_ipl has to be determined or did you stumble on "calibrate_by_bus_timing" yourself?

dakarym, 2017-08-23:
i haven't been back to think for a long while.  still something that is
interesting to do.  i'll have to put the hardware back together and get
back to it to try.  i'll see maybe this weekend

mursel, 2018-02-01:
As I understood I can use OneWireHub to make my attiny85 (in my case) as 1-wire device through emulation. Correct me if im wrong.

Giermann, 2018-08-29:
I'm using this library for almost 3 years now, but a rather old version, slightly patched to work on (my) ATtinys. I was unable to use more recent code on my 1-Wire bus because of the timing. Further I had to run the Tiny85 at 8 MHz, which I'm fine with - but still my clients keep to disappear from the bus, cause read errors - but everything turns good itself on the next reading, so I do not care much about it.

These days I started another approach to emulate a DS18B20 and faced the same problems as @dakarym .
I even tried to run the Tiny85 at 16MHz (PLL), which does a better job - but still the device disappears, gives read errors or simply 0xFF register content. I verified this with a Logic Analyzer, but don't know the source of the problem.
I ran "calibrate_by_bus_timing", which gave me value_ipl=13 at 8 and 16 MHz.
There's no bus issue, as I'm running a DS9490R in a test environment with less than 1m cable.
Everything works as expected on an Arduino Nano (at 8MHz, but not on the Tiny.

I read about the inaccurate clock timing of the Tinys and guess the Nano runs a real oscillator.
Does anybody have experience with this on Tiny85?

@mursel - at what environment does your attiny85 run perfectly?
Internal clock? Which bus master?
Maybe I should switch to my own (Arduino based) bus master to solve the timing issues on my production bus, but what's wrong with these Tinys?

Lukkoz, 2019-05-15:
I just tried the newest version of the library with ATtiny85. At first, it was now working at all but after switching from 1 MHz internal oscillator to 16 MHz external one everything works perfectly.