orgua/OneWireHub #31: hub.attach(ds2408) fails if OneWire master is live and USE_GPIO_DEBUG

ecnelson, 2017-04-24:
Again in the ds2408 example. The slave (OneWireHub) is connected to a master (OneWire) that is looping over a search for ds2408s. After a reset, the slave fails to initialize *unless* the OneWire pin is disconnected for a moment. The workaround was to turn off USE_GPIO_DEBUG in OneWireHub_config.h

The offending piece of code seems to be `waitLoops1ms()`.

orgua, 2017-04-24:
you are right, gpio-debug config shouldn't be activated by default. i will set a warning in the next release. i will leave this open till then

orgua, 2017-08-20:
ok, warning for "dangerous" constexpr-states is impossible right now, so i included some more static_asserts() to help to prevent enabled debug-output

closed 2017-08-20 by orgua