orgua/OneWireHub #34: Master unable to read emulated DS2431

Drevos, 2017-06-19:
This library seems perfect for what I'm trying to do, I have an actual DS2431 EEPROM chip and a master that can read everything off of the chip. For testing purposes I want to be able to emulate the chip and send data straight from my Arduino Uno for my master to read. 

However even though the master works with a real chip I'm not seeing any sort of data with the DS 2431 example program. I tried it with both pins 8 and 2 (Interrupt pin for the Arduino Uno) but nothing! The libraries appear to be in order and the program compiles but looking at the data line all I see are 0xFF...

Drevos, 2017-06-20:
So here's some things I've checked and troubleshooted-

-Arduino is up to date (v1.8.3)
-Using the latest library
-Clock speed on Arduino looks good with blinking program
-Arduino is powered, pin 8 connected to data line of master, master has an internal parasitic resistor that works for the physical chip (albeit I think its only 1k), ground is properly grounded
-Tried 9600 baud rate
-Tried enabling overdrive 

Looking at the data lines with Logicport I see this for the physical chip.
[image]

Then with the emulated chip I see this which appears to show a delay and shorter down times...

[image]

And again all I see are 0xFF over the data line.

orgua, 2017-06-23:
- can you tell us something about your master? 
- what are these pictures showing? A Failing data transmission? Would be interesting to see at least a reset and presence detection of the physical chip
- with some luck we find a simple timing problem in your logic-sniff and you just have to adjust timings in OneWireHub_config.h
- have you tried turning off USE_SERIAL_DEBUG in OneWireHub_config.h?
- there are people that have problems with the newest arduino version >1.8.2, but i haven't tested yet
- as noted in ds2431.h: datasheet is fuzzy, but device is similar to ds2433

Drevos, 2017-06-23:
- Sorry can't provide many details about the master aside from the fact that it works with real chips (I've used different DS431's with different values stored and it works just fine). 

- Both pictures are showing the 1-wire line as the master reads them, top is the successful physical read, bottom is the emulated. 

- I was hoping the same, I went and tried changing some values myself going off of the DS2431 data sheet but didn't get very far with that. 

- Isn't USE_SERIAL_DEBUG disabled by default? 

- I may try using a downgraded version of Arduino but likely won't get to working on this again until Monday.

However here's some addition pictures:

Master pulse when the line is empty/nothing is connected:
[image]

A closer look at the real chip being read.
[image]

A closer look at what the emulated chip is doing.
[image]

Drevos, 2017-06-28:
Tried Arduino 1.8.2, same results. Here's some much better Logic captures: 

Real chip holding 0's.
[image]

Emulated.
[image]

orgua, 2017-07-01:
your pictures and writings are confusing. picture above is the real chip, in overdrive, both traces the same. picture below is non-OD. 
These pictures don't help anyway. 
is there any way to provide or lend a ds2431? I am living in germany. 
Otherwise it would be good if you provide a data-log that captures normal onewire-operation, a bus search and a read and write of the eprom / ds2431. is your software freely available and allows capturing to file? i would need these logs with the following config: 
- original ds2431
- emulated ds2431 with serial- and gpio-debug disabled (even disable serial.begin() in setup-routine)
- emulated ds2431 with just gpio-debug enabled AND captured

the gpio-debug helps with pin-changes during important onewire-state-changes but may screw up overdrive-timing on a normal UNO.

Do you have a teensy at your disposal? These are fast enough to allow OD and GPIO-debug and even serial-output.

and at last you could check for me if original ds2431 still works if you have the UNO emulating another ds2431 (don't use the same address) on the same bus. Did you try to emulate any other devices? Maybe one that does not allow OD?

orgua, 2017-07-01:
oh, and i read your thread in the arduino-forum. send- and read-routines are ok. code has been pushed to the max to allow OD, it looks weird but it's fine. interrupts are always activated after the byte-operations are done.

Drevos, 2017-07-05:
I meant in each picture there are two traces however the bottom trace in each picture shows overdrive. Anyways I can see about getting the files of the traces soon however the chips aren't mine to lend unfortunately. 

I've only got the Arduino so I can't use the teensy, sorry. By same address do you mean same pin? I've tried a few other devices but I think my master only sends EEPROM specific commands so its hard to tell if they're functioning. 

Actually I realized I didn't set up Serial Debug to print properly. It's now showing the "Error: slave was used in incorrect way [0x3C]" Error after running the program which would explain the 1's. What exactly is meant by this error?

orgua, 2017-07-05:
i can't work or help you like that. 
- do you notice that both traces show the same plot? maybe OD is enabled, but it is not used, so it is irrelevant
- some basic understanding of onewire would tell you that there is an address for each device on the bus, like here: DS2431(DS2431::family_code, 0x00, 0x00, 0x31, 0x24, 0xDA, 0x00);
- your error message seems to be meaningless, because it is not pointing to the real problem. there should be NO debug-output enabled. your master seems to rely on overdrive. so disable all overhead, even throw out everything active with "serial". at least serial.begin(). then there's a chance to it to work with OD
- since you don't want to share information about your OW-master you are mostly on your own. Could be a timing-problem, could be everything else
- please try to follow the other given steps, otherwise i am out of ideas

Drevos, 2017-07-05:
Okay I was only double checking that you were referring to the family code/serial number because I've already tried altering that. I wanted to match it with a chip that I have but wasn't sure how when there are only 7 values and the real chip has 8. I believe that has something to do with CRC but that's another thing I was still trying to figure out.

Well I realized after commenting that the error code seems to be pointing at the 0x3C command being the issue, which is the Overdrive-Skip ROM command. I've got Overdrive_enable set to 1 and did away with all "serials" aside from .begin() with the same result.

orgua, 2017-08-20:
were you able to resolve your problems? 

i would close this for now, due to missing information. your issue could be a hint for timing-problems with your master-device. but we can't know for sure

vilplus, 2019-03-18:
Hi, I do not speak English well, but I will use a translator.
I have the same problem, I load the ds2431 emulator in arduino.
I use the program for flashing cartridge chips from Videojet.
When I read this program, everything is read, without naming the values ​​in the program code.
But it is necessary to connect the emulator to a real machine and start reading that problem is reading the data. The machine receives a serial number, but cannot read the data.
I can share the programs that I am trying to read the emulator to identify the problem and achieve the goal.

vilplus, 2019-03-18:
[1-Wire.zip](https://github.com/orgua/OneWireHub/files/2977474/1-Wire.zip)

vilplus, 2019-03-18:
[image]

vilplus, 2019-03-18:
Now I will read with a real ds2431 chip...

vilplus, 2019-03-18:
[image]

vilplus, 2019-03-18:
And this is another program that I can transfer only by mail. If you remove the verification data in the code, then you can program any data into it. But there is one thing. Now I will show in the picture.

vilplus, 2019-03-18:
[image]

vilplus, 2019-03-18:
Please help with project data.
It seems to me that the problem is in synchronization or with timings.
No data output. But suddenly there is a new program. Floating error.

vilplus, 2019-03-18:
After a little sitting in the program I found an interesting pattern.
When the emulator is fully read, it gives errors, but if I do everything in turn, everything reads.
If I count ROM then he recognizes it.
If I do skip ROM, then it reads the data.
I don’t know how to implement joint reading.

vilplus, 2019-03-18:
[image]

vilplus, 2019-03-18:
I will wait for help =)

orgua, 2019-03-18:
Looks like a classic timing problem with your master. There is a dedicated help-section on the main-page https://github.com/orgua/OneWireHub, after you worked through all these steps and the problem still persists you should ask in a user-forum. If that does not help you can come here to chat with the devs. I'm sorry, but my time is limited and i won't do your job for you. Onewire is hard on timing-constraints, but you have to be willing to do your share. There are a bunch of variables only you know about. 
In Short: Try to measure the timings of your master, or calibrate your slave accordingly. Or if you like to poke in the dark you can alter the VALUE_IPL number for your platform in platform.h.

tchiobtcha, 2025-01-09:
> I will wait for help =)

hello there 
im facing the same problem you had 
and also for flashing cartridge chips from Videojet.
i will be very grateful if you can update me how you solve this problem
im facing a huge problem in the factory where im working and this is my last ressort