orgua/OneWireHub #35: Master LinkUSB problem.

strike56, 2017-07-02:
Hi.

 In first, thank you very much for this library!

 On master DS9490R all work fine, without problem.
 But today I change my master to LinkUSB... And unfortunately all arduino devices disappeared from network (real 1wire devices still present without problem).

 Where may be problem??

orgua, 2017-07-02:
Some masters are not timing-conform. most ow-sensors / devices do not rely on this strict timing defined by maxim. so they seem to work better. 
i have already eased these rules, but need feedback from you. i don't own a linkUSB. is it possible to get a capture of the data line with the real sensors talking?

strike56, 2017-07-04:
Unfortunately I can't  capture real sensors talking.
But I make some debuging response. I change function "recvBit" (add debug points)
----------------------------------
bool OneWireHub::recvBit(void)
{
	Serial.println("-----");
	Serial.println(od_mode);
    noInterrupts();
    // Wait for bus to rise HIGH, signaling end of last timeslot
    timeOW_t retries = ONEWIRE_TIME_SLOT_MAX[od_mode];
		Serial.println(retries);
    while (!(DIRECT_READ(pin_baseReg, pin_bitMask)) && (--retries));
		Serial.println(retries);
    if (!retries)
    {
       _error = Error::RESET_IN_PROGRESS;
        interrupts();
        return true;
    };

    // Wait for bus to fall LOW, start of new timeslot
    retries = ONEWIRE_TIME_MSG_HIGH_TIMEOUT;
		Serial.println(retries);
    while ((DIRECT_READ(pin_baseReg, pin_bitMask)) && (--retries));
		Serial.println(retries);
    if (!retries)
    {
        _error = Error::AWAIT_TIMESLOT_TIMEOUT_HIGH;
        interrupts();
        return true;
    };

    // wait a specific time to do a read (data is valid by then), // first difference to inner-loop of write()
    retries = ONEWIRE_TIME_READ_MIN[od_mode];
		Serial.println(retries);
    while (!(DIRECT_READ(pin_baseReg, pin_bitMask)) && (--retries));
		Serial.println(retries);

    return (retries > 0);
};
-------------------------------------
And here is responses:

From DS9490R (example):
-----
0
166
166
18461
18384
24
24
-----

And from LinkUSB (always):
-----
0
166
166
18461
0

orgua, 2017-08-20:
did you make any progress with your problem?

orgua, 2017-08-20:
seems to be related to #23 
try to 10x the value ONEWIRE_TIME_MSG_HIGH_TIMEOUT in OneWireHub_config.h

strike56, 2017-08-22:
Ooooo YEEESS !!!
I set this value to 150000_us. And now all work fine on LinkUSB too...
Thank you very match! :-)

orgua, 2017-08-22:
Nice to know! 
If you ever have the chance to capture the communication with a logic analyser please let me know.

For documentation - I won't increase this value by default, because: 
- yes, this behaviour is allowed for OW, but rather unusual
- this will very much increase the time the controller spends in hub.poll(), often without reason. and leaves less time to do other things
- i made a note in the readme under section "help"

closed 2017-08-22 by orgua