orgua/OneWireHub #39: Emulate 2 DS18b20

NickCoolasia, 2017-07-11:
Hi, I when I try to emulate 2 ds18b20 my master couldn't read anything but when I emulate 1 ds18b20 with 1 ds18s20 my master can read the reading.

Is there any way to emulate 2 ds18b20 as there is always a 0.7 difference between ds18b20 and ds18s20?
Thank you.
Regards,
Nick

orgua, 2017-07-11:
so you say you still have your gps-master and it works with sensor A and sensor B separate, but when both are on the bus this experiment fails? Both have individual bus-addresses? seems that your master does not support "search rom" command and just uses "skip rom". so it does not have to discover the divices. Just adresssing "that" one device on the bus.

NickCoolasia, 2017-07-11:
No, when I attach ds18b20 and ds18s20 at the same time my master was able to read the reading. But when I try to attach 2 ds18b20 instead of 1 ds18b20 and 1 ds18s20, my master couldtn't read anything. 
[image]
[image]

orgua, 2017-07-11:
very odd, indeed. This example with 2x ds18b20 and 1x 18s20 does not have any problems. Do you use the newest library 2.1.1 and have every debug-code disabled in config? 

```
/*
 *    Example-Code that emulates a DS18B20
 *
 *    Tested with:
 *    - https://github.com/PaulStoffregen/OneWire --> DS18x20-Example, atmega328@16MHz as Slave
 *    - DS9490R-Master, atmega328@16MHz and teensy3.2@96MHz as Slave
 */

#include "OneWireHub.h"
#include "DS18B20.h"  // Digital Thermometer, 12bit

constexpr uint8_t pin_led       { 13 };
constexpr uint8_t pin_onewire   { 8 };

auto hub    = OneWireHub(pin_onewire);

auto ds18b20 = DS18B20(DS18B20::family_code, 0x00, 0x00, 0xB2, 0x18, 0xDA, 0x00); // DS18B20: 9-12bit, -55 -  +85 degC
auto ds18s20 = DS18B20(0x10, 0x00, 0x00, 0xA2, 0x18, 0xDA, 0x00);                 // DS18S20: 9   bit, -55 -  +85 degC
auto ds1822  = DS18B20(DS18B20::family_code, 0x00, 0x00, 0x22, 0x18, 0xDA, 0x00);                 // DS1822:  9-12bit, -55 - +125 degC

bool blinking(void);

void setup()
{
    Serial.begin(115200);
    Serial.println("OneWire-Hub DS18B20 Temperature-Sensor");
    Serial.flush();

    pinMode(pin_led, OUTPUT);

    // Setup OneWire
    hub.attach(ds18b20);
    hub.attach(ds18s20);
    hub.attach(ds1822);

    // Test-Cases: the following code is just to show basic functions, can be removed any time
    Serial.print("Test - set Temperatures to -56 degC (out of range): ");
    ds18b20.setTemperature(int8_t(-56));
    Serial.println(ds18b20.getTemperature());

    Serial.print("Test - set Temperatures to -55 degC: ");
    ds18b20.setTemperature(int8_t(-55));
    ds18s20.setTemperature(int8_t(-55));
    Serial.print(ds18b20.getTemperature());
    Serial.print(", ");
    Serial.println(ds18s20.getTemperature());   // ds18s20 is limited to signed 9bit, so it could behave different

    Serial.print("Test - set Temperatures to 0 degC: ");
    ds18b20.setTemperature(int8_t(0));
    ds18s20.setTemperature(int8_t(0));
    Serial.print(ds18b20.getTemperature());
    Serial.print(", ");
    Serial.println(ds18s20.getTemperature());

    Serial.print("Test - set Temperatures to 21 degC: ");
    const int8_t temperature = 21;
    ds18b20.setTemperature(temperature);
    ds18s20.setTemperature(temperature);
    ds1822.setTemperature(temperature);
    Serial.print(ds18b20.getTemperature());
    Serial.print(", ");
    Serial.println(ds18s20.getTemperature());

    Serial.print("Test - set Temperatures to 85 degC: ");
    ds18b20.setTemperature(int8_t(85));
    ds18s20.setTemperature(int8_t(85));
    Serial.print(ds18b20.getTemperature());
    Serial.print(", ");
    Serial.println(ds18s20.getTemperature());

    Serial.print("Test - set Temperatures to 126 degC (out of range): ");
    ds1822.setTemperature(int8_t(126));
    Serial.println(ds1822.getTemperature());

    Serial.println("config done");
}

void loop()
{
    // following function must be called periodically
    hub.poll();
    // this part is just for debugging (USE_SERIAL_DEBUG in OneWire.h must be enabled for output)
    if (hub.hasError()) hub.printError();

    // Blink triggers the state-change
    if (blinking())
    {
        // Set temp
        static float temperature = 20.0;
        temperature += 0.1;
        if (temperature > 30.0) temperature = 20.0;
        ds18b20.setTemperature(temperature);
        ds18s20.setTemperature(temperature);
        ds1822.setTemperature(temperature);
        Serial.println(temperature);
    }
}

bool blinking(void)
{
    constexpr  uint32_t interval    = 1000;          // interval at which to blink (milliseconds)
    static uint32_t nextMillis  = millis();     // will store next time LED will updated

    if (millis() > nextMillis)
    {
        nextMillis += interval;             // save the next time you blinked the LED
        static uint8_t ledState = LOW;      // ledState used to set the LED
        if (ledState == LOW)    ledState = HIGH;
        else                    ledState = LOW;
        digitalWrite(pin_led, ledState);
        return 1;
    }
    return 0;
}
```

NickCoolasia, 2017-07-11:
[image]
This is the response of my master. My server can show reading of ds18b20 and ds18s20 but there is always a difference of 0.7 between ds18b20 and ds18s20. 
I think this is the newest library because I just downloaded it yesterday

orgua, 2017-07-11:
Can't confirm your results. I extended the ds18b20-example to use ds18b20, ds18s20 and ds1822, twice each. setTemperature() was fed with 10 and 30 °C for these twins in int8 and float. and my ds9490-master showed the right temperatures every time. the master and the software-tool are both from maxim/dallas, so they should know their stuff. And i double-checked my emulation code twice. it is easy to get the datasheet wrong, because they do some silly stuff there. 
i'm afraid as long as you can't provide a log from a logic-analyzer you are on your own. 

```
auto ds18b20 = DS18B20(DS18B20::family_code, 0x00, 0x00, 0xB2, 0x18, 0xDA, 0x00);
auto ds18b21 = DS18B20(DS18B20::family_code, 0x00, 0x01, 0xB2, 0x18, 0xDA, 0x00);
auto ds18s20 = DS18B20(0x10, 0x00, 0x00, 0xA2, 0x18, 0xDA, 0x00);                 
auto ds18s21 = DS18B20(0x10, 0x00, 0x01, 0xA2, 0x18, 0xDA, 0x00);            
auto ds1822A  = DS18B20(0x22, 0x00, 0x0A, 0x22, 0x18, 0xDA, 0x00);  
auto ds1822B  = DS18B20(0x22, 0x00, 0x0B, 0x22, 0x18, 0xDA, 0x00);
bool blinking(void);

void setup()
{
    hub.attach(ds18b20);
    hub.attach(ds18b21);
    hub.attach(ds18s20);
    hub.attach(ds18s21);
    hub.attach(ds1822A);
    hub.attach(ds1822B);

    constexpr float tempA = 10;
    constexpr float tempB = 30;

    ds18b20.setTemperature(tempA);
    ds18b21.setTemperature(tempB);

    ds18s20.setTemperature(tempA);
    ds18s21.setTemperature(tempB);

    ds1822A.setTemperature(tempA);
    ds1822B.setTemperature(tempB);
}
```

ShanmukhaGanesh, 2017-07-15:
Hi guys,

When i'm connecting the emulator to a teltonika device its show an error of300 on the server side,
can anyone help me with this isuue.
Thanks,
ShanmukhaGanesh

orgua, 2017-07-15:
@ShanmukhaGanesh - not with this little information. Please work at least through the start page of the project. Especially section "HELP"

ShanmukhaGanesh, 2017-07-16:
@orgua - While I'm connecting the ds18b20 to a teltonika device(GPS TRACKER) directly it is able to transmit the data to my server.
Now I'm trying to use an arduino and emulate ds18b20 and connect it to my GPS tracker(teltonika) but it is not able to transmit the temperature reading to my server.
As I'm trying to make this wireless I'm using a wireless transmitter and a receiver the data from the transmitter is being transmitted on the receiver which is visible on the serial monitor in Arduino IDE but it is not able to send the data to teltonika which has a 1-wire port to accept the data from ds18b20.
Kindly, help me with this issue.

orgua, 2017-07-16:
sorry, a basic set of embedded skills on your side is needed to solve this problem. if you have serious effort, you would have a logic analyzer at hand. your problem sounds like a timing-issue. impossible to solve without a timing-analysis.  
and i think it would be easier to send the wireless received data directly via usb/serial to your server. onewire not needed

closed 2017-08-20 by orgua