orgua/OneWireHub #43: Master unable to read emulated DS2438

drtrigon, 2017-08-22:
I tested the DS2438 from M-o-a-T [1] to find that this is actually a fork of the code here. So I also tested this [2] and have the same issue that I cannot read from the sensor.

[1] https://github.com/M-o-a-T/OneWireHub/blob/master/examples/DS2438_battMon/DS2438_battMon.ino
[2] https://github.com/orgua/OneWireHub/blob/master/examples/DS2438_battMon/DS2438_battMon.ino

Details:
- code used was the DS2438_battMon on an Arduino Uno board (R3)
- master is a LinkHubE (iButtonLink) [3] emulating a DS2480B/9097U interface
- software used was owfs v3.1p0 on a raspberry pi

Issue: 
- owfs does recognize the sensor and lists it but cannot read from it (listed in owdir, error in owread)
  "ServerRead: Data error on 26.00003824DA00/temperature"

[3] https://cdn.shopify.com/s/files/1/0164/3524/files/The_LINK_Family_Manual_1.7.pdf?7145028656254187831 e.g. pages 10 and 11

orgua, 2017-08-22:
just a happy guess: maybe your problem is connected to issue #35 and #23? 
and please try the "help"-section in the the [readme](https://github.com/orgua/OneWireHub/blob/master/README.md) first. It was heavily extended since v.2.0-fork from m-o-a-t.
unfortunately i have no linkHubE to test it myself

orgua, 2017-08-22:
this [linkHubE](https://www.ibuttonlink.com/products/linkhube) costs 240 bucks? 
Would be nice to know if it uses a ft232R like the [linkUSB](https://www.ibuttonlink.com/products/linkusb). Do you mind opening the case? maybe there is an additional FT4232H with a quad serial. 

Both seem to support DS9097U emulation and can be used by the same software. so it could be. i should extend the help section and warn more about certain masters and offer solutions.  
 
if the timing adaptation works it would be nice if you narrow the value down a bit more. Try half of the value, and so on.

drtrigon, 2017-08-23:
Thanks for the answers! I am keen on testing this as soon as possible, but might take some days.

Yes I payed about 240 euros for that guy, but it is actually quite good and claims to do adaptive timing based on the line length, impedance, etc. I cannot promise anything, don't know how the case is closed (hope for screws and not glue), but I am willing to open it if possible.

Will report back my findings.

drtrigon, 2017-08-25:
Your "happy guess" solved my issue almost! (Thanks a lot!) I can work with it now but some of the values still cannot be read by owfs. Sometimes they are humidity, temperature and vis sometimes VAD and humidity.
I was not yet able to pin down the exact factor that is needed. I played with 2x, 5x and the suggested 10x all work better than the default value, but none solves all reading errors.

Writing pages (0-7) one at a time does work well but writing all pages at the same time does not work.

On the Arduino/AVR side I can read pages from 1 to 7 but I am not able to read and output the data stored in page 0.

drtrigon, 2017-08-26:
I switched to DS2431 as my intention is to build a LCD/OLED for 1wire. The memory is more suited. Is it normal that I have to read the memory after writing in order to trigger an update?

Will move on to the hardware build phase now, and then consider to open my master. Meanwhile I found [1] where is stated:

- Based on ibutonlink.com's LINK protocol.
- Uses a telnet interface to the LINK
- Internally
  - Serial LINK
  - 4 electrically separate (logically combined) arms
  - Lantronix Xport provides the network interface with web and telnet configuration
  - Default: Web configuration port 9999, Link at port 10001, DHCP client for network address

[1] http://owfs.org/index.php?page=linkhub-e

Generally I wanted to thank you for this really, really, really great library - I was looking for something like that for years! Thanks - great job! :)

orgua, 2017-08-28:
i have seen this behaviour with the failing long writes / reads before. there are some more timing-issues on your side. good news: they are fixable. a scope or logic analyser would help a lot. but here are possible fixes to try without knowing what is going on:
- some masters ignore the time-frame of the bit-slots, so you have to increase the ONEWIRE_TIME_SLOT_MAX value step by step, maybe around 5-10% for each test. 1 or 2 steps should be enough
- some masters have a different timebase. I just guess that yours is slower. so you could try to modify the value_ipl parameter in parameter.h a bit. 1 or 2 decrements should be enough, so 11 or 12 seems reasonable. you will notice that the successful transmission gets longer.

drtrigon, 2017-08-28:
Actually I found that your timings are very robust. My code was bad in terms of every loop with update (blinking() == True) took way too long. And still it was kind of usable! Now I changed basically to only take a lot of time if the whole display changes (e.g. initial write). This initial write then kind of makes the device freeze, but as soon as I read the memory the device starts running (looping and updating) again. So I think this is a good explanation for the behaviour. For sure it makes me happy as your code is robust and in cases it has issues (due to bad timing of MY additional code), it even recovers... very nice!

I do have an (old) analog scope which is not suited for these things, I would love to buy a logic-8 and thus am looking around for a good price (~ 500 bucks is too much). I would like to look bit more into these timings (and other stuff). The scope needs to be linux compatible ;))

When changing timing values how do you determine whether the communication becomes "better" or "worse"? Do you take statistics or is a scope needed for that?

orgua, 2017-08-29:
the saleae-brothers give you student-discount. or maybe you can buy a used old logic 8 or 16port. the first generation should be available for <100 bucks. chinesium knockoffs come for about 5$. the software is a blast! 
and for the timings: earlier in the development i tried to write big chunks of memory, around 32 byte on some devices. the initial values were 0xFF. when writing something to these banks with a bad timing there were only some of the first of the 32 bytes written (to something different than 0xFF). with better timing-values the success rate was getting higher and higher. 
i thought this could be a test for you too.

drtrigon, 2017-09-02:
Yes this sound like an "easy" test, will give that a try!
Thanks for the hints, will try to ask for a discount this is a good idea - "forgot" to think about this, however I don't think they give a 50% or more discount... ;))
Today I tested the DS2438 on an Arduino Nano, with factor 10x it basically works BUT again I have a lot "Error: Invalid argument" when trying to read e.g. voltages. Actually only the temperature works sometimes. This was done on an other master, an OW-SERVER [1].

[1] http://www.embeddeddatasystems.com/OW-SERVER-1-Wire-to-Ethernet-Server-Revision-2_p_152.html

Currently I cannot test on Arduino Uno as I used my last one (need to reorder) for this: https://www.instructables.com/id/Arduino-1-wire-Display-144-Chars/ :))