orgua/OneWireHub #47: Error: Invalid argument in OWFS

palsbo, 2017-10-19:
I am using an ATtiny85 to emulate DS2502.
It works fine with OWFS on Raspberry Pi
[image]
When using Synology DS115 (Same compiled version as on Pi), I get an error: "Invalid argument"
[image]
I think it is a timing issue, but can I fix it?

palsbo, 2017-10-19:
Ups! I realise that I am using an I2C to OW interface on the Pi but a DS9490 USB plug on the Synology.
Using the DS9490 on the Pi gives the same problem - Invalid argument!

I tried with several DS9390 but with the same result.

So - problem changed but still there!

orgua, 2017-10-27:
have you worked through the help section of the readme.md in the root-folder? 
propably the problem is connected to issue #43, #35, #23.
or maybe just #25

pklarsjo, 2018-07-23:
Thanks for a great library!

Any progress here? I am using attiny85 and experience the same for each command sent to DS2413. DS2405 is working fine, but I need readback possibility. DS2408 is also working fine with OWFS but is not working well with Domoticz, leaving me with DS2413 as the last option for controlling outputs. So, is there something that I need to try to change, given the fact that with 05 and 08 the timing seems ok?

(Master is a ds2482-800 on RPI)

orgua, 2018-07-23:
@pklarsjo so did you work through my previous tipps? it seems to be that timing-issue. easy to fix by changing one value

pklarsjo, 2018-07-23:
No, not yet since timing is ok with other types. But maybe i need to try
differenting them?

orgua, 2018-07-23:
of course you should! ds2405/2413 does almost nothing on the bus, very short messages. when more memory is involved the ow-messages get longer. timing problems are more likely to occur.

pklarsjo, 2018-07-23:
Ok, will try and come back. Which exact parameter do you suggest changing
and to what (currently no scope available)?

pklarsjo, 2018-07-24:
ONEWIRE_TIME_MSG_HIGH_TIMEOUT changed to x10, no difference. Some additional info:
Exact command:  echo '1' > PIO.A
Reading from device is no problem:  cat sensed.A

My code:
/*
 *    Example-Code that emulates a DS2413 Dual channel addressable switch
 *
 *    Tested with
 *    - DS9490R-Master, atmega328@16MHz and teensy3.2@96MHz as Slave
 */

#include "OneWireHub.h"
#include "DS2413.h"  // Dual channel addressable switch

constexpr uint8_t pin_led       { 1 };
constexpr uint8_t pin_onewire   { 0 };

auto hub    = OneWireHub(pin_onewire);
auto ds2413 = DS2413( DS2413::family_code, 0x00, 0x00, 0x13, 0x24, 0xDA, 0x00 );    // Work - Dual channel addressable switch

bool blinking(void);

void setup()
{
    pinMode(pin_led, OUTPUT);

    // Setup OneWire
    hub.attach(ds2413);
}

void loop()
{
    // following function must be called periodically
    hub.poll();

    // Blink triggers the state-change
    if (blinking())
    {
                digitalWrite(pin_led, ds2413.getPinState(0));
    }
}

bool blinking(void)
{
    const  uint32_t interval    = 1000;          // interval at which to blink (milliseconds)
    static uint32_t nextMillis  = millis();     // will store next time LED will updated

    if (millis() > nextMillis)
    {
        nextMillis += interval;             // save the next time you blinked the LED
        static uint8_t ledState = LOW;      // ledState used to set the LED
        if (ledState == LOW)    ledState = HIGH;
        else                    ledState = LOW;
        return 1;
    }
    return 0;
}

pklarsjo, 2018-07-26:
This is getting even more strange. I enabled toggling of the state in the blinking function (ds2413.setPinState(0,ledState);) to see if device is live or not. After some normal 1sec flashes it suddenly freezes on or off. I know for sure it's not rebooting since I have added some fast boot blinks, it just stays for a while longer than expected.

In addition I've tried changing some parameters withou getting any progress (ONEWIRE_TIME_MSG_HIGH_TIMEOUT * 10, ONEWIRE_TIME_SLOT_MAX (120/135/150), VALUE_IPL (12, 13, 14) without any improvements. 

Starting to despair... Is someone else using this device; has it been verified?