orgua/OneWireHub #54: DS2430 support

alex-byi, 2018-04-04:
Hello. What is about DS2430 support? Is it hard to implement this?
We tried to do it by coping DS2431 and modify in accordance with specs.
https://github.com/Neschur/OneWireHub/commit/4714adc1d9fc93fd582c9d48391e692d771ad301

We achieved correct work with arduino EPROM reader(https://github.com/pceric/DallasEPROM), but our real device doesn't work with emulator, it is only able to read serial number.

Could you review our commit?

alex-byi, 2018-04-06:
This is our progress https://github.com/orgua/OneWireHub/commit/5c225e51f3f70f22ad38b2a189891a9062146af9
We achieved some success, all is work, except read memory. Maybe it's a bug, because Search signal and read memory signal have same code - 0XF0, And it 0XF0 doesn't appear in duty method.

We will continues work

orgua, 2018-07-23:
ok, you seemed to made progress on that, #55
bus-cmds and device-cmds are on two different levels, they can be the same, no interference. 
any more questions?