orgua/OneWireHub #56: DS2405 on attiny85 with Domoticz

pklarsjo, 2018-04-16:
Hi, 

trying to make DS2405 work with Domoticz. Problem is that it ds2405 example (modified) changes state on positive flanks (in order to set a digital 1 and then 0 on led pin requires putting PIO to 1-0-1. Probably this is correct according to DS2405 datasheet but it does not work well with Domoticz implementation. Any suggestions on how to solve this?

My modified DS2405 example:
```
#include "OneWireHub.h"
#include "DS2405.h"  // switch
constexpr uint8_t pin_led       { 1 };
constexpr uint8_t pin_onewire   { 0 };
constexpr uint8_t addr[6]= {0x01, 0x00, 0x00, 0x00, 0x00, 0x00};

auto hub    = OneWireHub(pin_onewire);
auto ds2405 = DS2405( DS2405::family_code, addr[0], addr[1], addr[2], addr[3], addr[4], addr[5]);    

void setup()
{
    uint8_t i;
    pinMode(pin_led, OUTPUT);
    hub.attach(ds2405);
    ds2405.setPinState(0);
}

void loop()
{
    // following function must be called periodically
    hub.poll();
	//Keep busy doing nothing
    doSomething(); 
    digitalWrite(pin_led, ds2405.getPinState());
}

bool doSomething(void)
{
    const  uint32_t interval    = 1000;          
    static uint32_t nextMillis  = millis();     

    if (millis() > nextMillis)
    {
        nextMillis += interval;             
        return 1;
    }
    return 0;
}
```

orgua, 2018-07-23:
so this does not seem hub-related, right? what you do with the pin-state after reception should be part of your own implementation. otherwise i don't understand your problem.

pklarsjo, 2018-07-23:
You are right, the implementation of ds2405 is correct; it should always
toggle. Saw that now in the datasheet. I need to go with another variant
(ds2413 or 08).

orgua, 2018-07-23:
yeah, this thing is very limit. and ds2405.h also states: 
// works, but reading back the value is not supported, because alarm search is not implemented yet

closed 2018-07-23 by orgua