orgua/OneWireHub #58: Unpowered device breaks bus...

codmpm, 2018-05-23:
First, thank you for this superb library 👍

Currently we are building a soil-moisture sensor which will be released as open-source. It emulates the DS2438 and reads the temperature from an attached DS18B20 as well as "borrowing" the serial number from it to get a unique device id. We ware also using the DS2438's memory to store configuration data and are working on a configuration interface based on an ESP8266 to configure the devices with a smartphone over onewire.

[image]

We are currently in the prototype phase and everything works out great. But, we are stuck with one problem:

When we cut the 5V line of the sensor (slave) the bus breaks down, meaning we can't read data anymore. 
We are testing with our ESP8266 implementation and an 3,3 to 5V [level shifter based on a BS170](https://wiki.fhem.de/wiki/1-Wire_Pegelwandler). The sensor (slave) uses pin A0 of the ATMega328p and is directly wired to the onewire bus.

My first thought was the internal resistence of the ATmega when it's unpowered, but this should be high enough to not interfere with the pull-up for the onewire line of the master (2k2@5V).

Do you have any suggestion how to separate the onewire slave - meaning ATMega328p - when it's not powered? Did we miss something on the slave connecting the bus?

Would be great to have your input on this.

Cheers,
Patrik

pklarsjo, 2018-06-28:
Interesting project! Remember reading somewhere that emulated devices can't use databus for supply as the original devices can, it simply consumes too much power.

codmpm, 2018-06-29:
Hey @pklarsjo,

we do not power the MCU over the 1-wire bus itself, there's always a dedicated 5V line for this. 
The Problem seems, that the ATMega tries to power itself over the data line if either GND or VCC is missing.

We sepearate the MCU from the 1-wure bus by using a [MAX9940](https://datasheets.maximintegrated.com/en/ds/MAX9940.pdf) now. This will not bring down the bus if GND or VCC on the MCU are missing. We've found only some rare circumstances where it will happen now.

Neverthelesse we think it's a good idea to seperate the bus from the MCU and protect the inputs with the 9940 and give it ome protection.

gahujipo, 2018-08-29:
@codmpm have you been able to resolve your issue?

codmpm, 2018-08-29:
As I did not get another suggestion we went with the MAX9940. I'm open to any other idea.

taliesin, 2019-04-20:
Probably this is way too late, but maybe others with the same issue find this helpful.

I use a BSS159 (and a ESD-Diode to GND directly on the connector, which is not in the schematic) to protect my 1-wire slave pins (left is bus input, right is micro controller pin).

This will prevent overvoltage on the micro controller (I'm using STM32F070 with 5V tolerant pins here) and will make the pin high impedance (above about 2V) if the micro controller power (+3.3V) is off.
3.3V is generated from +5V bus power using linear voltage regulator with some ceramic capacitors on the output (a few 100nF).

[image]

Extremely slow ramping inputs might slowly charge the 3.3V, but with the standard 1-wire setup this usually works.