orgua/OneWireHub #60: DigiSpark Attiny85 fix for compiling errors due to missing gnu++11 flag

git-Space, 2018-06-04:
Compiling can fail with all sorts of errors, due to a missing **-std=gnu++11** flag in:

> "C:\Users\%username%\AppData\Local\Arduino15\packages\digispark\hardware\avr\1.0.0\platform.txt"

I got it working modifying **compiler.cpp.flags=**  by adding the missing flag at the end of the list, like so:

`compiler.cpp.flags=-c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -MMD  -std=gnu++11`

After this, everything is fine !

> Sketch uses 5910 bytes (98%) of program storage space. Maximum is 6012 bytes.
> Global variables use 181 bytes of dynamic memory.
> Running Digispark Uploader...
> Plug in device now... (will timeout in 60 seconds)

By the way, I am running the Attiny85 slave on pin 0 (contrary to the suggested pin 2, because I needed all the A0-A3):
```
/*   4CH ADC over 1-Wire  
          +-DigiSpark-----------------+
          |  __                       |
          | /  \               A0 [o] |1 P5
          | \__/               A2 [o]~|3 P4 USB-
         ++-----+              A3 [o] |2 P3 USB+
         | uUSB |  i0 SCK  SCL A1 [o]~|7 P2 USCK
         ++--__-+     MISO        [ ]~|6 P1
          | /  \      MOSI SDA    [o]~|5 P0 AREF
          | \__/  [o][o][ ]           | 
          +------------------Attiny85-+          
                  5V GND Vin(6-16V)
*/

constexpr uint8_t pin_onewire   { 0 }; //0 is P0, 1 is P1
```
and it worked well, 4x DS18B20 are received on the master which is a NANO running [orgua's OneWire.h](https://github.com/orgua/OneWire) (but I still can't get the DS2450 to work without a CRC error, neither on the NANO as slave nor the Attiny85 as slave, and I am not sure if it's due to the CRC calculation itself or a bus timing issue - logic analyser should arrive soon :champagne: ).

In summary, I fixed my compiling issues by adding the above mentioned flag.
Oh, and OneWireHub is an awesome library!

orgua, 2018-07-23:
thanks for the detailed log! 
it is odd that this crc is failing, i have tested these examples with a lot of combinations. did your logic analyser help?

mariostjr, 2018-09-10:
I got same errors here with AtmelStudio 7 and ATMEGA328PB.  By adding thoses flags, most errors got solved, but im still getting
"

> invalid conversion from 'unsigned char' to 'unsigned char*' [-fpermissive]

"

To code: 
```template<typename T1>
T1 * portInputRegister(const T1 port) { return port; };
```

My pin declaration:

`constexpr uint8_t pin_onewire   { 4 };`

joachim198, 2019-09-11:
Hi! Thanks for the info! Now programming works. 
But I have an issue when my Digispark Attiny85 starts up while connected to a second arduino uno with the "onewire". If I unplug the cable and start the Attiny, he boots correctly and then I can plug the "onewire" into the uno. Has anyboy else this issue? Even programming the Attiny over usb while the onewire is connected won't work? Any idea?