orgua/OneWireHub #61: CRC failure in DS2408

sabas1080, 2018-06-05:
Hi @orgua 

I have been able to work the example of the DS18b20 in a SAMD21 with the following modifications
https://github.com/orgua/OneWireHub/pull/59

but I get the following error when trying the example of DS2408

`CRC failure in DS2408 at 2900000824DA008A`

orgua, 2018-07-23:
i need more information! please work through the help-section of the readme.md, especially the last point

papvlieg, 2021-11-16:
Hi @orgua,

I have the same problem as described by sabas1080. I connected two arduino nano's (pin D10 to pin D8 with 4k7 pullup resistor). They are at close distance on a breadboard.

The first acts as a master using the OneWire lib example: DS2408_Switch, the second as a DS2408 using the OneWireHub lib example.  The output of the master is (each second):

  Reading DS2408 2900000824DA008A
CRC failure in DS2408 at 2900000824DA008A
No more addresses.

I also tried to disable the CRC check but the 8 DS2408 pin states are all 1, while pin state 3 should be 0 according to the example code (and the serial output shows that it is zero).

What am I doing wrong?

closed 2025-09-01 by sabas1080