orgua/OneWireHub #62: Onewirehub devices crashes all onewirenet after some time

pklarsjo, 2018-07-23:
I have noticed that after a few hours/days uptime on my emulated devices (and all my "real" devices) are lost in OWFS. They simply are no longer listed, and a power toggle of the master RPI is needed to get back up again. Could this be timing related? In my net I have DS2890 and some of DS2405/08/13 connected. 

Running on attiny85 and DS2482-800 as master on an RPI.

orgua, 2018-07-23:
are only the real devices gone or the emulated ones also?

this deadlock can be everything
- there could be memory leak that slowly fills the ram, depending on your program, i think the lib is safe (but can't promise that)
- some arduino-fn that does not behave the way it should. the ports for the attiny are not very well written 
- millis() and micros() shouldn't be the culprit, one overflows at 70mins, the other after 52 days

pklarsjo, 2018-07-23:
All decices, both real and emulated.

I now noticed that +5v supply to the pi and slaves is to low; 4.63V. Will
correct that and see if it helps.

orgua, 2018-07-23:
4.6 should be still enough. but yeah, you never know what cheap voltage regulators were used. 
some more questions for you: 
- can you still access the Pi after that deadlock? 
- do you have some kind of keep-alive led on your attiny? 1s blinks are very helpful!

pklarsjo, 2018-07-23:
Yes, pi is still alive. Listing /mnt/1wire shows some basic folders like
bus.0..7 etc.  but device  folders are gone. Reboot of pi does not solve
this, but a power toggle does.

Nope, no life sign led. I can correct this.

orgua, 2018-07-23:
ok, so your attiny really dies somehow. The emulated IC's are harmless and don't need many resources. memory allocation is all static. Could you make your program public? you could also send it to my mail-address. i think you will find it. don't want to post it here

pklarsjo, 2018-07-23:
Well, my implementation is nothing fancy. Its a 2890 example that is
modified so that it sets wiper value to pwm. But I can post it later.

pklarsjo, 2018-07-24:
Here comes my code:
/*
 *    Example-Code that emulates a DS2890 Single channel digital potentiometer (datasheet has it covered for up to 4 CH)
 *
 *    Tested with
 *    - DS9490R-Master, atmega328@16MHz and teensy3.2@96MHz as Slave
 */

#include "OneWireHub.h"
#include "DS2890.h"  // Single channel digital potentiometer

constexpr uint8_t pin_led       { 1 };
constexpr uint8_t pin_onewire   { 2 };

auto hub    = OneWireHub(pin_onewire);
auto ds2890 = DS2890( 0x2C, 0x00, 0x00, 0x90, 0x28, 0xDA, 0x00 );    // Work - Single channel digital potentiometer

bool blinking(void);

void setup()
{
    pinMode(pin_led, OUTPUT);

    // Setup OneWire
    hub.attach(ds2890);
    analogWrite(pin_led,10);
}

void loop()
{
    // following function must be called periodically
    hub.poll();

    // Blink triggers the state-change
    if (blinking())
    {
        analogWrite(pin_led,ds2890.getPotentiometer(0));
    }
}

bool blinking(void)
{
    const  uint32_t interval    = 500;          // interval at which to blink (milliseconds)
    static uint32_t nextMillis  = millis();     // will store next time LED will updated

    if (millis() > nextMillis)
    {
        nextMillis += interval;             // save the next time you blinked the LED
        static uint8_t ledState = LOW;      // ledState used to set the LED
        if (ledState == LOW)    ledState = HIGH;
        else                    ledState = LOW;
        //digitalWrite(pin_led, ledState);
        return 1;
    }
    return 0;
}

orgua, 2018-07-24:
so this exact code hangs after some days on an attiny? Very odd. Which attiny-lib are you using? is it https://github.com/damellis/attiny ? Can you try it on a bigger chip like teensy or atmega328?