orgua/OneWireHub #63: No presence after an aborted Search ROM

kutukvpavel, 2018-08-13:
Hi,
I am trying to develop a DS1990A emulator that is supposed to work with some generic access control systems and I've stumbled upon two issues which original 1990-s do not exhibit.

The first one is, the library does not generate a presence pulse when the slave device is first connected to the bus. Obviously, master (that does not have to continuously poll the line) does not know that a device has been connected and therefore does not start communication. Of course this issue can be solved pretty easily by monitoring the bus "manually" prior to calling hub.poll(), but as an essential part of any ds2401-like slave's functionality it really should be a part of the lib. Also very long low pulses should be probably considered a disconnect from the bus... And a pulldown resistor (I am using 100k) has to be present to avoid floating. Not sure how to implement it though, use RESET_TOO_LONG maybe.

As for the second one... I've come across a master device that issues only two timeslots after sending 0xF0 (Search ROM) command and then aborts the transaction issuing a reset (see attachment). Of course, the master expects a presence pulse to follow that reset, but OneWireHub does not generate it (at all, with out-of-the-box timings). Consequently, master restarts communication process again and again thinking that the device is disconnected every time. I've created a clone of that master using Arduino Nano and I've been able to reproduce the issue right away (and perform some tests).
While I was searching for a solution I found out that I was using an old version of the library (2.0.0) and I upgraded it to the latest, however it didn't help.
Using v 2.0.0 I've been able to make presence pulse appear occasionally (almost 50/50 probability ratio) by playing with timings (i reckon I mostly reduced timeslot length and increased write zero time, tweaked reset/presence length just a little bit), but when I updated the lib config was overwritten and I never managed to get presence again.
In 2.0.0 the problem was somehow dependent on previous transactions. My particular master sends three commands in the following order: 0x33, 0x03, 0xF0. The problem disappeared when I changed the order to 0x03, 0x33, 0xF0. Yes, 0x03 is a non-standard one and my slave is not supposed to answer it at all, its purpose is to identify rewritable (and non-finalized) devices such as RW1990 (for them 0x03 and 0x33 are identical). Once again, for latest version order makes no difference - the issue just persists, no matter what.
 
I've looked through the library code and found no obvious flaws: the workflow seems completely legit. SearchIDTree exits with RESET_IN_PROGRESS error when comes across a reset pulse instead of a timeslot, then poll starts again from the beginning, i.e. from reset detection. But I can't understand why reset detection fails. Is time consumed somewhere and the pulse becomes too short to be identified as reset? My code basically calls only hub.poll() inside loop() and performs a single direct pin reading, so delays between polls are almost as short as possible. Only one DS2401 is attached to the hub. No serial or debug whatsoever. Calibration seems to be spot-on, ONEWIRE_TIME_WRITE_ZERO remains within 1uS tolerance according to logic analyzer.
I'm running Arduino Nano @ 16MHz, arduino IDE v1.8.5.
Here is some logic analyzer data related to the second issue. Green is genuine, blue is emulated. Look at the last pulse (and its absence).
[image]
Overall communication structure remains the same.
[image]

kutukvpavel, 2018-08-13:
Well, attempt to modify recvBit() failed. OW_T_READ_MIN_TO_MAX is constexpr-ed READ_MAX - READ_MIN.
`// wait a specific time to do a read (data is valid by then)
	retries = ONEWIRE_TIME_READ_MIN[od_mode];
	while ((DIRECT_READ(pin_baseReg, pin_bitMask) == 0) && (--retries != 0));
	if (retries > 0) return true; //Exit if 1 was received

	//Otherwise check pulse length to detect possible reset
	retries = OW_T_READ_MIN_TO_MAX[od_mode];
	while ((DIRECT_READ(pin_baseReg, pin_bitMask) == 0) && (--retries != 0));
	if (retries > 0) return false; //Return if OK

	//Return error if the pulse is too long
	_error = Error::RESET_IN_PROGRESS;
	return true;
`
With this code hub no longer responds to any commands but reset, though it adds about 15uS of execution time only.
Obviously, the quick and dirty solution now (with line 283 commented out) is to decrease RESET_MIN value, but I'll try to find better one.

joysfera, 2019-01-23:
@kutukvpavel have you got it working reliably at last? I'm facing serious troubles when the OneWireHub is to share the bus with real DS18B20. The timing in OneWireHub_config.h seems completely off and it kinda blocks the real DS18B20 due to too long reset pulse (or what).

Even the author's comments in the following code seems like they have no idea what's going on:
```
constexpr timeOW_t ONEWIRE_TIME_PRESENCE_MIN[2]      = {   160_us,  8_us }; // was 125
constexpr timeOW_t ONEWIRE_TIME_PRESENCE_MAX[2]      = {   480_us, 32_us }; // should be 280, was 480
```

kutukvpavel, 2019-01-25:
@joysfera Indeed I have. Though I did not experience any major issues with timings and common 1-wire communication. My problems were related to not-so-common usage scenarios (tricky master device and having to handle connection/disconnection from the bus). As for the timings, they worked out-of-the-box, despite being a bit off (and I don't understand the comments too).
Overall I don't find this lib itself buggy of unreliable, it's actually quite a good one for common applications such as ds18.
Therefore I suspect that your issues have simpler causes than mine. Let's try to figure them out, starting from some obvious stuff.
1. What device are you using? Is it clocked right?
2. Are you using example ds18b20 thermometer code or some custom stuff?
3. When exactly the problem arises (when emulated DS is connected together with a genuine one, or even when emulated DS is alone on the bus etc)?
4. What have you tried so far? Do you have a logic analyzer?

joysfera, 2019-01-25:
@kutukvpavel thanks for your interest. For emulating DS18B20 slave I use "Pro Mini", an arduino clone with ATMEGA328P @ 16.000000 MHz.

The problem arises when emulated DS18B20 is connected together with recent buggy/fake/counterfeit Chinese DS18B20s that don't cope with the unprecise timing. Specifically it seems (judging from my logical analyzer sleepless nights) that the too long PRESENCE_MIN time causes the Chinese fake DS18B20 to reset or something. Then they start communicating in the middle of OneWireHub talking and the whole bus goes to hell.

Unfortunately noone but me yet proved that the Chinese DS18B20 are buggy counterfeits so whenever you connect these together with the emulated ones the suspicion falls onto the emulated one.

Anyway, the OneWireHub is also to blame, because even though the PRESENCE_MIN is set to 160 us (and the comment right next to it says "was 125 us") it actually generates pulse long more than 272 us (which is out of spec: 60 to 240 us). So there's something fishy in the library as well. I have yet to figure it out where the additional 112 us is spent...

I tried shortening the PRESENCE_MIN and _MAX times (because the _MAX time is also weird - 480 us with comment "was 280 us"), got it working more reliably at first but it seems that in some extended periods of time (say a week or two weeks) something bad happens and the library stops communicating altogether (not happened to me yet, but others using the emulated device reported it to me).

So all in all, the library seems OKish at first but together with other real devices on the same bus, especially with Chinese DS18B20, does not work at all or causes weird issues.

kutukvpavel, 2019-01-25:
@joysfera 
>it actually generates pulse more than 272 us long
Ok, then I assume the problem is due to emulator. No 1-wire device can be expected to work flawlessly side-by-side with this, chinese clones are not to blame.
Is that value at least consistent? What about other timings, are they off too?
The lib measures time in while-loop-cycles. Since I know that my copy produces values inside a margin of error (let's say 5%), the cause must be related to external interference. Two things are straightforward to blame: calibration parameters and use of interrupts (showPresence() does not disable them).
I suppose you didn't change VALUE_IPL for AVR, but it relies on microsecondsToClockCycles(), which relies on F_CPU. Is your build environment standard arduino IDE?
Since you have a logic analyzer, have you tried calibration sketch from examples with GPIO-debug enabled? Is 1mS really 1mS?

Btw, just looked through config and noticed:
//  should be --> datasheet
//  was       --> shagrat-legacy
"was" relates to values that existed long time ago before forking ancient source from another person's repo. God knows why his timings were so strange back then, but nowadays it has nothing to do with our problems.