orgua/OneWireHub #65: Loxone and Wemos D1

gahujipo, 2018-08-20:
I tried to emulate a DS2438 on a Wemos D1 and find the "sensor" with Loxone. The Wemos D1 has only 3,3V pins so I used a [level shifter](https://www.watterott.com/index.php?page=product&info=2861) to rise the signal to 5V.  But it couldn't be found. 

My wiring scheme is this: 
[image]

My current code is this 

```Cpp
#include "OneWireHub.h"
#include "DS2438.h"

constexpr uint8_t pin_led{LED_BUILTIN};
constexpr uint8_t pin_onewire{0}; // Arduino D3

auto hub = OneWireHub(pin_onewire);
auto ds2438 = DS2438(DS2438::family_code, 0x00, 0x00, 0x38, 0x24, 0xDA, 0x00); // todo what hex code for serial number

// the setup function runs once when you press reset or power the board
void setup()
{
	Serial.begin(115200);
	Serial.println("OneWire-Hub DS2438 BME680");

	pinMode(pin_led, OUTPUT);

	// Setup OneWire
	hub.attach(ds2438);

	Serial.print("Test: set Temperature in float 38 deg C: ");
	ds2438.setTemperature(38.0f);  // can vary from -55 to 125deg
	Serial.println(ds2438.getTemperature());
}

// the loop function runs over and over again until power down or reset
void loop()
{
	// following function must be called periodically
	hub.poll();

	// Blink triggers the state-change
	if (blinking())
	{
		static float    temp = 10.0;
		static uint16_t volt_10mV = 10;
		static uint16_t current = 10;

		if ((temp += 0.05) > 30.0) temp = 10.0;
		if ((volt_10mV++) > 200) volt_10mV = 10;
		if ((current++)   > 200) current = 10;

		ds2438.setTemperature(temp);
		ds2438.setVoltage(volt_10mV);
		ds2438.setCurrent(current);

		Serial.println(temp);
	}
}

bool blinking(void)
{
	const  uint32_t interval = 2000;          // interval at which to blink (milliseconds)
	static uint32_t nextMillis = millis();     // will store next time LED will updated

	if (millis() > nextMillis)
	{
		nextMillis += interval;             // save the next time you blinked the LED
		static uint8_t ledState = LOW;      // ledState used to set the LED
		if (ledState == LOW)    ledState = HIGH;
		else                    ledState = LOW;
		digitalWrite(pin_led, ledState);
		return 1;
	}
	return 0;
}
```

Is this correct or am I missing something?

gahujipo, 2018-09-06:
I followed the steps in the README and set up an emulated DS18B20 which works, even more than 8 hours in line. After that I was even able to stabilise the communication so that my master was able to find DS2438. I was able to get data from the emulated DS2438 once, but after that my master isn't able to get further data. The DS2438 remains discoverable, but doesn't return any data. I tested this with an interval of 30 seconds between each read cycle. Do you have a hint what I could try else?

gahujipo, 2018-09-07:
After adjusting the following values 
- lowering this one step to
```C++
constexpr uint8_t VALUE_IPL { (microsecondsToClockCycles(1) > 120) ? 25 : 21 }; // instructions per loop
```
- and adjusting this to
```C++
constexpr timeOW_t ONEWIRE_TIME_SLOT_MAX[2]          = {   148_us, 42_us }; // should be 120, measured from falling edge to next falling edge 
```
like mentioned here: https://github.com/orgua/OneWireHub/issues/43#issuecomment-325297441
I am able to communicate for aprox. three hours with the emulated sensor where every second operation fails with an poll interval of 25sec. 
Has anyone further suggestions to stabilise the bus?

raintonr, 2020-12-26:
> The DS2438 remains discoverable, but doesn't return any data.

I see exactly the same thing here with Loxone system.

I discovered a different version of DS2438. The code wasn't on Github but I copied it for a very basic MQ135 sensor:

https://github.com/raintonr/MQ135As1W

This works perfectly. Been stable here for months.

However, I recently wanted to do something more complex involving using `setVADVoltage` and `setVDDVoltage` (new methods in code linked above). When trying to use them both they often get reported as zero for some reason. Will try and figure that out.

raintonr, 2020-12-27:
On this topic, I found the following very useful page describing DS2438 communications and the mechanism to read VAD or VDD:

https://eds.zendesk.com/hc/en-us/articles/214484863-Working-with-the-DS2438-using-the-HA7E-HA7S

When a DS2438 is configured in Loxone it always reads both VAD and VDD. It's true to say, that due to other limitations in Loxone (regarding number of sensors supported on a bus) it could be useful to have 4 values presented on a single sensor, so reading the extra voltage could be useful, but how many bytes on the wire does it 'cost' to get the single, extra 10 bit voltage value?

Looking at the above link, seems like 270-ish bytes (both TX/RX) are required to configure the A/D bit, initiate voltage conversion and read results (well - if I've understood it correctly, please tell me if that's not the case).

If that is correct, using a DS2438 sounds a terribly inefficient way to get just one extra 10 bit value and one would likely be much better off emulating multiple DS1822 (or similar) devices.

raintonr, 2020-12-28:
In case I didn't make it clear, PR #97 seems to fix the DS2438 problem with Loxone here.

Found this through trial and error seeing which of the changes in the modified version I linked above made things work. Just looks like it was because sometimes calcCRC was skipped & guess Loxone may be causing this skip, or actually checking CRCs where other masters don't?

raintonr, 2020-12-28:
3fc416dece2b5c1d18fb7d6adb46abc45840189b enables emulation of VDD/VAD voltage reading which Loxone users should find useful.

PR #97 (2022-10-08): #65 Break loop rather than return so drops through to calcCRC
Using a Loxone master, this change makes the sensor reliably appear. Before the change, sensor could be detected with a scan but never read. Fixes #65.
closed 2022-10-08 by orgua