orgua/OneWireHub #69: One hardware slave works, multiple hardware slaves work for a short time only

sebascarra, 2018-11-22:
Hello, I have the following setup:

[image]

(*)Code on master Arduino:

`#include <OneWire.h>

/* DS18S20 Temperature chip i/o */

OneWire  ds(10);  // on pin 10

void setup(void) {
  Serial.begin(19200);
}

void loop(void) {
  byte i;
  byte present = 0;
  byte data[12];
  byte addr[8];
  
  if ( !ds.search(addr)) {
    Serial.print("No more addresses.\n");
    ds.reset_search();
    delay(250);
    return;
  }
  
  Serial.print("R=");
  for( i = 0; i < 8; i++) {
    Serial.print(addr[i], HEX);
    Serial.print(" ");
  }

  if ( OneWire::crc8( addr, 7) != addr[7]) {
      Serial.print("CRC is not valid!\n");
      return;
  }
  
  if ( addr[0] != 0x28) {
      Serial.print("Device is not a DS18S20 family device.\n");
      return;
  }

  // The DallasTemperature library can do all this work for you!

  ds.reset();
  ds.select(addr);
  ds.write(0x44,1);         // start conversion, with parasite power on at the end
  
  delay(1000);     // maybe 750ms is enough, maybe not
  // we might do a ds.depower() here, but the reset will take care of it.
  
  present = ds.reset();
  ds.select(addr);    
  ds.write(0xBE);         // Read Scratchpad

  Serial.print("P=");
  Serial.print(present,HEX);
  Serial.print(" ");
  for ( i = 0; i < 9; i++) {           // we need 9 bytes
    data[i] = ds.read();
    Serial.print(data[i], HEX);
    Serial.print(" ");
  }
  Serial.print(" CRC=");
  Serial.print( OneWire::crc8( data, 8), HEX);
  Serial.println();
}`

This setup works for a while. TX blinks about once every second or so, both on the master Arduino as on the slaves. The serial monitor reports 22 devices (7 Nano 1 + 7 Nano 2 + 7 Nano 3 + real sensor). Eventually, after 1 minute or so, one of the slave Arduinos fail. The pin 13 LED blinks fast, and its associated slaves fail to appear on the serial monitor). Eventually, all slaves fail, it's kind of random but I can say one fails every 1-2 minutes. This is all set up on a protoboard just in case you ask.

What I actually want to do is:

I need to create a multidrop network with 1 slave (an Arduino Nano running this library) simulating 1 sensor, one slave every 50 cm along a Cat5e cable, for 100 m. That's 200 simulated slaves.

I have a crappy USB oscilloscope (Hantek 6022BE) I might be able to use to debug, but do you think I will be able to run 200 slaves using this? Or I might be better with real Dallas devices instead?

Thank you & regards.

sebascarra, 2018-11-29:
Alright guys, I've checked with a cheap USB logic analyzer, and I highly suspect that my issue is related to this:

https://github.com/arduino/Arduino/issues/4492

However, I did not investigate further. I ended up using the library OneWireArduinoSlave which is not as capable as this one, but it does what I need (emulate just one device per slave) and is very compact and easy to edit (just one cpp and one header file).

Anyways, I hope me or someone else can investigate this issue further. Maybe it's a memory leak? A reset wouldn't reset the microcontroller, I had to power it down.

Regards

thomasesr, 2019-06-24:
Did you figure out the problem yet? I'm having the same issue. I have my one wire on pin 8 with 4.7k pull-up resistor and LED on 13. Whenever I connect another device on the same 1-wire line, either a Hardware DS18b20 or another Arduino nano it stops recognizing the arduinos.

sebascarra, 2019-06-24:
No, I've had to migrate to this library: https://github.com/neuoy/OneWireArduinoSlave

However, I could do it because I only needed one software slave per hardware slave in the end. So you might be out of luck. This is something that requires deeper investigation.