orgua/OneWireHub #75: All  not work (arduino 1.8.5)

selevo, 2019-04-28:
i tested all  examples
 nothing responsde and not detected in  'OneWireViewer' software of dallas
(OneWireViewer version 3.19.47) But hee detect real devices.

i  start Test-Code for calibration
nothing response. LED on 13pin all time on
serial port  all time : OneWire-Hub calibration by observing the OW-Bus

What problem ?
This project worked ?

selevo, 2019-04-30:
log data trasfer 
[image]

selevo, 2019-04-30:
data log trasfer
https://www.youtube.com/watch?v=NzzGL_rsOjA&feature=youtu.be

selevo, 2019-05-02:
ds18b20 OneWireHub-0.2.0 worked in all rduino's with gcc 4.8.1
and not worked with arduino's +gcc4.9.2 (it's arduino version 1.6.10 and up)

ebelouet, 2019-08-23:
HI
I try with gcc 4.8.1 and i have device in list but after we can't read any temperature and i have crc error