orgua/OneWireHub #80: Does not compile for arduino zero

milesfrankland, 2019-09-02:
These are the errors:

```
OneWireItem.h:4: In file included from
BAE910.h:8: from
BAE910.cpp:1: from
 
OneWireHub.h: In function constexpr timeOW_t operator"" _us(long long unsigned int)
 
OneWireHub.h: 13:1: error: the value of 'SystemCoreClock' is not usable in a constant expression
 
samd21g18a.h:239: In file included from
samd21.h:69: from
samd.h:105: from
sam.h:470: from
Arduino.h:48: from
platform.h:8: from
OneWireHub.h:4: from
OneWireItem.h:4: from
BAE910.h:8: from
BAE910.cpp:1: from
system_samd21.h:53: note  uint32_t SystemCoreClock is not const
   extern uint32_t SystemCoreClock;   \*!< System Clock Frequency (Core Clock)  *
 
OneWireItem.h:4: In file included from
BAE910.h:8: from
BAE910.cpp:1: from
 
OneWireHub.h: In function constexpr timeOW_t timeUsToLoops(uint16_t)
 
OneWireHub.h: 20:1: error: the value of 'SystemCoreClock' is not usable in a constant expression
 
samd21g18a.h:239: In file included from
samd21.h:69: from
samd.h:105: from
sam.h:470: from
Arduino.h:48: from
platform.h:8: from
OneWireHub.h:4: from
OneWireItem.h:4: from
BAE910.h:8: from
BAE910.cpp:1: from
system_samd21.h:53: note  uint32_t SystemCoreClock is not const
   extern uint32_t SystemCoreClock;   \*!< System Clock Frequency (Core Clock)  *
 
OneWireHub.h:22: In file included from
OneWireItem.h:4: from
BAE910.h:8: from
BAE910.cpp:1: from
OneWireHub_config.h: At global scope
```

Compiles fine for AVR stuff though!

orgua, 2019-12-29:
yeah, this is fucked up. my current solution is a hack - faster uCs should not need the loop-counting. I hope i find some time in the near future to update this lib.

kiowadriver, 2022-07-12:
The same issue with the Trinket M0 (SAMD21E18A).  The onewire lib as well as this OneWireHub lib both indicate compatibility with the __SAMD21G18A__.  

I found this [pull reques](https://github.com/adafruit/MAX31850_OneWire/issues/5)t for the OneWire library that seemed to resolve some compatibility issues but unfortunately more errors still persist after implementing the same fix.

`Compiling .pio\build\adafruit_trinket_m0\lib9b2\OneWireHub\DS2431.cpp.o
In file included from .pio\libdeps\adafruit_trinket_m0\OneWireHub\src\OneWireItem.h:4,
                 from .pio\libdeps\adafruit_trinket_m0\OneWireHub\src\BAE910.h:8,
                 from .pio\libdeps\adafruit_trinket_m0\OneWireHub\src\BAE910.cpp:1:
.pio\libdeps\adafruit_trinket_m0\OneWireHub\src\OneWireHub.h: In function 'constexpr timeOW_t operator""_us(long long unsigned int)':
.pio\libdeps\adafruit_trinket_m0\OneWireHub\src\OneWireHub.h:13:1: error: the value of 'SystemCoreClock' is not usable in a constant expression
   13 | }
      | ^`

I can't seem to find a better library for slave emulation but all seem to be out of date. What am I missing? Has something better than OneWire come along?

netmilk, 2023-07-31:
Hey 👋, 

I have the same issue with the **STM32F0** board when trying to emulate the DS2431. Thanks for all the work so far! I'd love to help because this sits on the critical path for my application, but I'm not proficient in C++. Let me know if I can assist in any other way, please.

Cheers!
A

```
In file included from Documents/Arduino/libraries/OneWireHub/src/OneWireItem.h:4,
                 from Documents/Arduino/libraries/OneWireHub/src/DS2431.h:9,
                 from /private/var/folders/0h/tvr3mlyx0wl3rbpljhckvy_h0000gn/T/.arduinoIDE-unsaved2023631-2135-1t8d9vt.d3dzg/sketch_jul31a/sketch_jul31a.ino:9:
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h: In function 'constexpr timeOW_t operator""_us(long long unsigned int)':
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:13:1: error: the value of 'SystemCoreClock' is not usable in a constant expression
   13 | }
      | ^
In file included from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/stm32f030x6.h:103,
                 from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/stm32f0xx.h:125,
                 from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/stm32_def.h:28,
                 from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/stm32/clock.h:19,
                 from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/wiring_time.h:23,
                 from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/wiring.h:38,
                 from Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/cores/arduino/Arduino.h:36,
                 from /private/var/folders/0h/tvr3mlyx0wl3rbpljhckvy_h0000gn/T/arduino/sketches/45D704DEBA72B1C0696EBD061C77E0F8/sketch/sketch_jul31a.ino.cpp:1:
Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/system_stm32f0xx.h:56:17: note: 'uint32_t SystemCoreClock' is not const
   56 | extern uint32_t SystemCoreClock;          /*!< System Clock Frequency (Core Clock) */
      |                 ^~~~~~~~~~~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h: In function 'constexpr timeOW_t timeUsToLoops(uint16_t)':
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:20:1: error: the value of 'SystemCoreClock' is not usable in a constant expression
   20 | }
      | ^
Library/Arduino15/packages/STMicroelectronics/hardware/stm32/2.5.0/system/Drivers/CMSIS/Device/ST/STM32F0xx/Include/system_stm32f0xx.h:56:17: note: 'uint32_t SystemCoreClock' is not const
   56 | extern uint32_t SystemCoreClock;          /*!< System Clock Frequency (Core Clock) */
      |                 ^~~~~~~~~~~~~~~
In file included from Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:22:
Documents/Arduino/libraries/OneWireHub/src/OneWireHub_config.h: At global scope:
Documents/Arduino/libraries/OneWireHub/src/OneWireHub_config.h:27:59: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
   27 | constexpr timeOW_t ONEWIRE_TIME_RESET_TIMEOUT        = {  5000_us };        // for not hanging to long in reset-detection, lower value is better for more responsive applications, but can miss resets
      |                                                           ^~~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:9:20: note: 'constexpr timeOW_t operator""_us(long long unsigned int)' declared here
    9 | constexpr timeOW_t operator "" _us(const unsigned long long int time_us) // user defined literal used in config
      |                    ^~~~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub_config.h:28:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
   28 | constexpr timeOW_t ONEWIRE_TIME_RESET_MIN[2]         = {   430_us, 48_us }; // should be 480
      |                                                            ^~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:9:20: note: 'constexpr timeOW_t operator""_us(long long unsigned int)' declared here
    9 | constexpr timeOW_t operator "" _us(const unsigned long long int time_us) // user defined literal used in config
      |                    ^~~~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub_config.h:29:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
   29 | constexpr timeOW_t ONEWIRE_TIME_RESET_MAX[2]         = {   960_us, 80_us }; // from ds2413
      |                                                            ^~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:9:20: note: 'constexpr timeOW_t operator""_us(long long unsigned int)' declared here
    9 | constexpr timeOW_t operator "" _us(const unsigned long long int time_us) // user defined literal used in config
      |                    ^~~~~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub_config.h:32:61: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
   32 | constexpr timeOW_t ONEWIRE_TIME_PRESENCE_TIMEOUT     = {    20_us };        // probe measures 25us, duration of high state between reset and presence
      |                                                             ^~~~~
[... cut ...]
   43 | constexpr timeOW_t ONEWIRE_TIME_WRITE_ZERO[2]        = {    30_us,  8_us }; // the hub holds a zero for this long
      |                                                             ^~~~~
Documents/Arduino/libraries/OneWireHub/src/OneWireHub.h:9:20: note: 'constexpr timeOW_t operator""_us(long long unsigned int)' declared here
    9 | constexpr timeOW_t operator "" _us(const unsigned long long int time_us) // user defined literal used in config
      |                    ^~~~~~~~

exit status 1

Compilation error: exit status 1
```

netmilk, 2023-07-31:
Do I understand correctly this issue is connected to this remark in the [docs](https://orgua.github.io/OneWireHub/features_hub.html)?

```
at the moment the lib relies sole on loop-counting for timing, no direct access to interrupt or timers, NOTE: if you use an uncalibrated architecture the compilation-process will fail with an error, look at ./examples/debug/calibrate_by_bus_timing for an explanation
```

jenslam, 2024-03-08:
I'm trying to include onewirehub into an existing ESP32 based PlatformIO project.
When adding the lib_deps for OnewireHub I'm getting errors in OneWireHubConfig.h in line 47 and following

`
constexpr timeOW_t ONEWIREHUB_TIME_RESET_TIMEOUT = {5000_us};
`

Error Message: 
`
constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
`
I'm not sure if this is OT, but it atleast seems to be an comparable issue..