orgua/OneWireHub #81: Slave on Digistump ATTiny85 (rev.3, china)

joachim198, 2019-09-14:
I had problems using onewirehub  on my chinese Digistump. Pin 3 & 4 were not working for onewire (USB ports). Pin 5 was configured as reset. (You can try to pulldown Pin5 with a 220Ohm resistor to ground, if the program stops "voila!" its a reset). Pin 0 and 2 are in usage with I2C over tinyWireM. So Pin1 was the only left, but it only worked when removing the led connected to Pin1. And now it works!
Thanks a lot for this library!
[main.txt](https://github.com/orgua/OneWireHub/files/3612558/main.txt)

eigenein, 2019-12-19:
Hi @joachim198,

I'm trying to implement a similar thingy and struggling with my master not being able to recognise my slave. Since you managed to get it working, could you share more details please?

- Which particular board frequency have you used?
- Is it USB-synced? Because I'm suspecting some timing issues
- You mentioned USB pins didn't work for you. Was it because of a live USB connection or disregardless of this?

orgua, 2019-12-29:
hi @eigenein: a forum would be a better starting-point for this kind of development, because it is not software-related. There is a general instruction on the main readme.md on https://github.com/orgua/OneWireHub and also some hints for attiny: use pin2 or pin3 for Attiny, source: https://github.com/gioblu/PJON/wiki/ATtiny-interfacing

closed 2019-12-29 by orgua