orgua/OneWireHub #82: Write protection after address 0x03 DS2502

AndiP1234, 2019-11-12:
const uint8_t reg_RA = translateRedirection(reg_TA[0]);

                if (getPageProtection(**reg_TA[0]**)) // reg_TA[0] must be replace with reg_RA

`
case 0x0F:      // WRITE MEMORY

            while (reg_TA[0] < sizeof_memory)
            {
                if (hub->recv(&data))       break;
                crc = crc8(&data,1,crc);

                if (hub->send(&crc))        break;

                const uint8_t reg_RA = translateRedirection(reg_TA[0]);

                if (getPageProtection(reg_TA[0]))
                {
                    const uint8_t mem_zero = 0x00; // send dummy data
                    if (hub->send(&mem_zero)) break;
                }
                else
                {
                    memory[reg_RA] &= data; // EPROM-Mode
                    setPageUsed(reg_RA);
                    if (hub->send(&memory[reg_RA])) break;
                }
                crc = ++reg_TA[0];
            }
            break;`

AndiP1234, 2019-11-14:
There is one more bug at
`if (getPageProtection(reg_TA[0]))`
you need to replace reg_TA[0] with page

complete fixed case:
```
case 0x0F:      // WRITE MEMORY

		while (reg_TA[0] < sizeof_memory)
		{
			if (hub->recv(&data))       break;
			crc = crc8(&data, 1, crc);

			if (hub->send(&crc))        break;

			const uint8_t reg_RA = translateRedirection(reg_TA[0]);
			const uint8_t page = static_cast<uint8_t>(reg_RA >> 5);
			if (getPageProtection(page))
			{
				const uint8_t mem_zero = 0x00; // send dummy data
				if (hub->send(&mem_zero)) break;
			}
			else
			{
				memory[reg_RA] &= data; // EPROM-Mode
				setPageUsed(page);
				if (hub->send(&memory[reg_RA])) break;
			}
			crc = ++reg_TA[0];
		}
		break;`

orgua, 2019-12-29:
thanks for spotting this thing! but just to be clear - you are talking of one bug, right? your second message implies another bug, but i think you are talking of a missing line of code (page-generator) and a wrongfully protection-check (reg_RA instead of page). Feel free to open again if i missed something

closed 2019-12-29 by orgua