orgua/OneWireHub #83: DS2438 Humidity

jrhilton, 2019-11-14:
The readme file has DS2438 in bold implying all functions are implemented, but what about setting humidity? Any plans to implement this?

oschade, 2019-12-28:
And in the docs is written "Smart Battery Monitor, measures temperature, **2x voltage** and current, 10bit".
But there is only one function "setVoltage" to set one voltage value.
Would be very great, if you could update this!

orgua, 2019-12-29:
@jrhilton: as you can see in the readme, the ds2438 has no humidity-function. There seems to be another IC connected to this 1Wire-Device that outputs voltage or currents. You have to figure it out yourself

orgua, 2019-12-29:
@oschade: please read the datasheet. there is one ADC and a control bit for an input selector. Data is always stored in the same register. You are right, i haven't implemented the check for the input-selector, but the use is very limited. If you need more storage, there are user-memory-pages 3 to 7 with 8 bytes each. `DS2438.writeMemory()` is public!

``

closed 2019-12-29 by orgua

raintonr, 2020-12-28:
Check out PR #97 which allows emulation of VDD/VAD voltage reading.