orgua/OneWireHub #84: Livolo thermostat  sensor emulation

eigenein, 2019-12-24:
Hi,

This is rather knowledge sharing because I found nothing over the Internet, but should you have some insights – this would be helpful for me.

I got a [Livolo thermostat](https://www.livolo.eu/c-4632899/livolo-thermostat/) which has an internal DS18B20. Since I already have an NTC thermistor mounted into my floor, I have soldered out the built-in Dallas and decided to make a NTC-to-OneWire adapter with a Chinese Arduino Nano board.

I have tested the thermostat with an external real DS18B20 and it worked fine.

The Livolo has pretty weird 1-Wire master implementation. I spent pretty much an entire weekend to fine-tune the timings and went to the following:

```cpp
// The master sends out too long reset signal, `35000_us` is the lowest value that worked.
constexpr timeOW_t ONEWIRE_TIME_RESET_MAX[2]         = {   50000_us, … };

// Otherwise, you get `AWAIT_TIMESLOT_TIMEOUT_HIGH` and/or `FIRST_BIT_OF_BYTE_TIMEOUT`
constexpr timeOW_t ONEWIRE_TIME_MSG_HIGH_TIMEOUT     = { 30000_us };

// Otherwise, you get `INCORRECT_ONEWIRE_CMD`. I assume because then it's read incorrectly.
constexpr timeOW_t ONEWIRE_TIME_READ_MIN[2]          = {    35_us,  … };
```

I also changed `hub->send(scratchpad, 9);` to `hub->send(scratchpad, 2);` since the thermostat only reads 2 bytes. Otherwise I get `FIRST_BIT_OF_BYTE_TIMEOUT`.

This gives correct and stable measurement displayed on the thermostat (except for rare `VERY_SHORT_RESET`s).

**And now the most weird part which I haven't solved yet.**

In exactly 3m 36s after I plug the thermostat into a wall socket something happens inside the Livolo. It starts to read a measurement in a stably incorrect way. Examples:

- It shows 6°C (`0b110`) instead of 10 (`0b1010`)
- 5/6/7 instead of 21 (`0b10101`)

So it looks like it misreads some bits. But there're a couple of things that totally confuse me:

- The thermostat reads out a measurement every 1 second. However, an incorrect value **is stable**. I mean, you plug it in, you see 21°C every second for 3m36s, and then you see e.g. 6°C _every second_. You plug it off, plug in again, you see 21°C again for 3m36s and then you see e.g. 5°C. So every second it makes a same error but the error is different every time you plug it into a socket.
- Looks like that every ~3-4 mins it "picks" a different error to make. E.g. 21°C for 3m36s, 5° for 3-4 mins, 6° for 3-4 mins, 7° for 3-4 mins and so on.
- If I plug a real DS18B20 in, it works just fine.
- If I restart my Nano, it still shows an incorrect value.

I tried to play with the other timings to no avail. Nothings changes besides that extreme values break readings at all. At the moment I'm kinda stuck and it maybe that I won't finish it ever. But the above might help other people.

eigenein, 2019-12-24:
Apparently, I managed to reproduce this with a real DS18B20, so the issue is no longer `OneWireHub`-specific. Looks like it's really a bug of the thermostat itself.

The trick is to connect a real Dallas, wait the magic 3m36s and then disconnect and connect again **only** the data wire. Somehow if disconnect the power wires too, the bug isn't reproduced. If I don't disconnect the real Dallas, the bug is also not reproduced.

So to summarise, at 3m36s uptime real Dallas survives something that is equal to re-connecting its data wire. And my Nano with `OneWireHub` doesn't survive the same. Looking further…

eigenein, 2019-12-24:
I got him.

It turned out that the thermostat is buggy. When I soldered the built-in Dallas back and turned it on, displayed value jumped from a correct one to a random one at 3m36s. Looks like a buggy firmware whatever. Gonna communicate that to the manufacturer.

closed 2019-12-24 by eigenein

tobiaswaldvogel, 2021-06-15:
As mentioned before the thermostat looks nice but the software is totally buggy.
I wrote a new firmware from scratch which addresses all this and much more.

[https://github.com/tobiaswaldvogel/livolo-thermostat](https://github.com/tobiaswaldvogel/livolo-thermostat)