orgua/OneWireHub #93: DS2431 Difference behavior between real eeprom and arduino emulator

x225am, 2020-12-06:
Hi all, I have some devices with DS2431 eeprom I have read all info from it  loaded in to the Arduino eeprom emulator. For some reason when I trying use this emulator in the real device it just would not accept it. I read with logic analayzer packets between master and eeprom and compare them with emulator. For some reason emulator would not respond on requests.
Please have a look screens from the Analayzer:
This is from original DS2431 eeprom
[image]
And this one is from emulated. For some reason it only reset packets.
[image]

orgua, 2020-12-06:
cool that you have the needed tools and interest in getting this to run. unfortunately your screenshot does not tell much. its 99% a timing-problem (as usual). Well, i can only give you the same answer as yesterday: please read through the main readme. There is a dedicated help-section

https://github.com/orgua/OneWireHub/blob/master/README.md

orgua, 2020-12-06:
(feel free to reopen if there is a real bug)

closed 2020-12-06 by orgua