orgua/OneWireHub #102: OneWireHub on ESP32 PROBLEM

lisdaniel, 2021-08-28:
Hello oruga,
hello dear community.

I've only had electronics and coding for a few days and this is my first post on GitHub.
I have already successfully simulated several OneWire devices on a MEGA328 (Arduino NANO). All examples could be compiled without problems.
When trying to compile the examples for the "NODE MCU ESP-32S" board, I get a lot of error messages from the Arduino IDE.
Can someone help me and tell me exactly what steps I need to take to use the OneWireHub on an ESP 32?

Thanks so much. I hope for your help!
Have a nice Weekend!!!

Here are the error messages when trying to compile the example: "DS18B20_thermometer" for the board: "NODEMCU ESP-32S":

_**In file included from C:\Users\Daniel\AppData\Local\Temp\arduino_build_369790\sketch\DS18B20_thermometer.ino.cpp:1:0:
D:\Arduino\libraries\OneWireHub\src/OneWireHub.h: In function 'constexpr timeOW_t operator""_us(long long unsigned int)':
C:\Users\Daniel\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32/Arduino.h:82:67: error: call to non-constexpr function 'uint32_t getCpuFrequencyMhz()'
 #define clockCyclesPerMicrosecond() ( (long int)getCpuFrequencyMhz() )
                                                                   ^
C:\Users\Daniel\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32/Arduino.h:84:46: note: in expansion of macro 'clockCyclesPerMicrosecond'
 #define microsecondsToClockCycles(a) ( (a) * clockCyclesPerMicrosecond() )
                                              ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub.h:11:31: note: in expansion of macro 'microsecondsToClockCycles'
     return timeOW_t(time_us * microsecondsToClockCycles(1) / VALUE_IPL); // note: microsecondsToClockCycles == speed in MHz....
                               ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub.h: In function 'constexpr timeOW_t timeUsToLoops(uint16_t)':
C:\Users\Daniel\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32/Arduino.h:82:67: error: call to non-constexpr function 'uint32_t getCpuFrequencyMhz()'
 #define clockCyclesPerMicrosecond() ( (long int)getCpuFrequencyMhz() )
                                                                   ^
C:\Users\Daniel\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc2\cores\esp32/Arduino.h:84:46: note: in expansion of macro 'clockCyclesPerMicrosecond'
 #define microsecondsToClockCycles(a) ( (a) * clockCyclesPerMicrosecond() )
                                              ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub.h:19:23: note: in expansion of macro 'microsecondsToClockCycles'
     return (time_us * microsecondsToClockCycles(1) / VALUE_IPL); // note: microsecondsToClockCycles == speed in MHz....
                       ^
In file included from D:\Arduino\libraries\OneWireHub\src/OneWireHub.h:22:0,
                 from D:\Arduino\libraries\OneWireHub\examples\DS18B20_thermometer\DS18B20_thermometer.ino:9:
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h: At global scope:
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:27:59: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_RESET_TIMEOUT        = {  5000_us };        // for not hanging to long in reset-detection, lower value is better for more responsive applications, but can miss resets
                                                           ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:28:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_RESET_MIN[2]         = {   430_us, 48_us }; // should be 480
                                                            ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:28:68: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_RESET_MIN[2]         = {   430_us, 48_us }; // should be 480
                                                                    ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:29:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_RESET_MAX[2]         = {   960_us, 80_us }; // from ds2413
                                                            ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:29:68: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_RESET_MAX[2]         = {   960_us, 80_us }; // from ds2413
                                                                    ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:32:61: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_PRESENCE_TIMEOUT     = {    20_us };        // probe measures 25us, duration of high state between reset and presence
                                                             ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:33:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_PRESENCE_MIN[2]      = {   160_us,  8_us }; // was 125
                                                            ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:33:69: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_PRESENCE_MIN[2]      = {   160_us,  8_us }; // was 125
                                                                     ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:34:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_PRESENCE_MAX[2]      = {   480_us, 32_us }; // should be 280, was 480
                                                            ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:34:68: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_PRESENCE_MAX[2]      = {   480_us, 32_us }; // should be 280, was 480
                                                                    ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:37:58: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_MSG_HIGH_TIMEOUT     = { 15000_us };        // there can be these inactive / high timeperiods after reset / presence, this value defines the timeout for these
                                                          ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:38:60: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_SLOT_MAX[2]          = {   135_us, 30_us }; // should be 120, measured from falling edge to next falling edge
                                                            ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:38:68: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_SLOT_MAX[2]          = {   135_us, 30_us }; // should be 120, measured from falling edge to next falling edge
                                                                    ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:41:61: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_READ_MIN[2]          = {    20_us,  4_us }; // should be 15, was 30, says when it is safe to read a valid bit
                                                             ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:41:69: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_READ_MIN[2]          = {    20_us,  4_us }; // should be 15, was 30, says when it is safe to read a valid bit
                                                                     ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:42:61: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
[... cut ...]
 constexpr timeOW_t ONEWIRE_TIME_WRITE_ZERO[2]        = {    30_us,  8_us }; // the hub holds a zero for this long
                                                             ^
D:\Arduino\libraries\OneWireHub\src/OneWireHub_config.h:43:69: error: 'constexpr timeOW_t operator""_us(long long unsigned int)' called in a constant expression
 constexpr timeOW_t ONEWIRE_TIME_WRITE_ZERO[2]        = {    30_us,  8_us }; // the hub holds a zero for this long
                                                                     ^
cc1plus.exe: warning: unrecognized command line option '-Wno-frame-address'**_

eychei, 2022-12-23:
Hi,

I did have the same issue and tried so many options to get this working.
The library just works fine with my esp8266 but not with my esp32.
Error was always:

error: call to non-constexpr function 'uint32_t getCpuFrequencyMhz()'

So looking into 
\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32\Arduino.h
and comparing to 
\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Arduino.h

I found the difference in the following line
esp8266:
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
esp32:
#define clockCyclesPerMicrosecond() ( (long int)getCpuFrequencyMhz() )

 and here is the SOLUTION:

Open File:
\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32\Arduino.h
change
#define clockCyclesPerMicrosecond() ( (long int)getCpuFrequencyMhz() )
to
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )

Thats it now it compiles without errors.

-e

GeorgeIoak, 2023-04-08:
I ran into this problem trying to get OneWireHub running on a ESP32. Although the change mentioned does allow the code to compile I cannot get it to work on a ESP32 (2 different boards). If I switch to an old Arduino board the code and hardware works as expected but I need OneWireHub running on a ESP32 for it's WiFi.

The Readme says this works with ESP32 but perhaps some changes have broken that ability?

svdrummer, 2023-04-08:
Which pins are you using

GeorgeIoak, 2023-04-08:
In the readme I saw this comment: 

> during hub-startup it issues a 1ms long high-state (you can check the instruction-per-loop-value for your architecture with this)

So I hooked up a scope and checked and I'm measuring a pulse at boot up of 1.3ms instead of 1ms with a ESP32 Dev Module:

[image]

I tried to run the calibration but that just kept throwing panic errors.

I have this hooked up to a master and I was under the impression that I should get more pulses but that's not happening, only the initial one during boot up.

GeorgeIoak, 2023-04-08:
Relying on the fact that the `GPIO_DEBUG_PIN` should go low for 1ms upon bootup I experimented with changing `VALUE_IPL` and found that by changing from the default of `39` to `51` I was able to dial the pulse width to 1.0ms:
[image]

The code is now working as expected on this old ESP32_DEVKIT_V1 module with flash frequency set to 80MHz and CPU frequency set to 240MHz in the Arduino IDE.

Leitner1, 2025-11-30:
I found the problem. I renamed all entries in the LIB files (OneWireHub\src) from "MEM_SIZE" to "OWH_MEM_SIZE". Now it compiles without errors.

Now I have another issue. When I enable "USE_GPIO_DEBUG", the code doesn't execute. I get no output in the serial console. The following is displayed on the debug pin:

1ms/div:
[image]

20ms/div:
[image]

Any idea?