orgua/OneWireHub #123: Support for Raspberry RP2040

ahorn42, 2023-05-21:
Hello,

has someone already tried to run OneWireHub on a Raspberry RP2040 like device? Funnily these µC are cheaper than most 1-wire devices, and I would like to use some RP2040 to fake a bunch of DS2450.

After renaming `MEM_SIZE` to `MEMORY_SIZE` (or something else, as `MEM_SIZE` seems to be a macro in the RP2040 arduino wrapper) I got the code compiling, but I could not get it working, so that the emulated 1-wire devices could be read from another µC.

I already found the `platform.h` file but I am not sure, how to adapt this to RP2040 and also I have no clue, if this is really the issue. Any ideas how to go on?

best regards,
ahorn

ahorn42, 2023-05-21:
Just found the `calibrate_by_bus_timing` script and it seems to work now with `IPL = 25` :)

Unfortunately I still don't understand what the other values mean that were configured in `platform.h`

```c
#define PIN_TO_BASEREG(pin)             (portInputRegister(digitalPinToPort(pin)))
#define PIN_TO_BITMASK(pin)             (digitalPinToBitMask(pin))
#define DIRECT_READ(base, mask)         (((*(base)) & (mask)) ? 1 : 0)
#define DIRECT_MODE_INPUT(base, mask)   ((*((base)+1)) &= ~(mask))
#define DIRECT_MODE_OUTPUT(base, mask)  ((*((base)+1)) |= (mask))
#define DIRECT_WRITE_LOW(base, mask)    ((*((base)+2)) &= ~(mask))
#define DIRECT_WRITE_HIGH(base, mask)   ((*((base)+2)) |= (mask))
using io_reg_t = uint8_t; // define special datatype for register-access
```

Above is the sample for `AVR`, I kind of have an idea what it should do, but I think I need some further hints how to adapt this to the RP2040 ;)

orgua, 2023-05-22:
That's mostly Atmel-related lowlevel-stuff to access pins that was needed to allow the fast overdrive-mode of Onewire. Maybe it's time to remove all that error-prone platform-glue and use arduino-commands by default. 
I will order a RP2040 and look into either porting the lib or advance to pure-arduino as a major v3-step

aa4649, 2023-05-25:
Hello all,
I am also trying to migrate.
But it seems to be a double definition error.

For some reason, it seems that MEM_SIZE has already been defined as 512 bytes, and an error occurs stating that it cannot be redefined to 144 bytes in the DS2431.h header.
--
https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json
"Raspberry Pi Pico /RP2040"

c:\Users\aaaa\Arduino\libraries\OneWireHub\src/DS2431.h:15:31: error: expected unqualified-id before numeric constant
   15 |     static constexpr uint8_t  MEM_SIZE          { 144 };
      |                               ^~~~~~~~

c:\Users\aaaa\Arduino\libraries\OneWireHub\src/DS2431.h:18:72: error: narrowing conversion of '512' from 'int' to 'uint8_t' {aka 'unsigned char'} [-Wnarrowing]
   18 |     static constexpr uint8_t  PAGE_COUNT        { MEM_SIZE / PAGE_SIZE };
      |                                                                        ^

exit status 1
--

#include "OneWireHub.h"
#include "DS2431.h"

constexpr uint8_t pin_onewire   { 2 };

auto hub = OneWireHub(pin_onewire);

auto ds2431 = DS2431(DS2431::family_code ,0x11 ,0x22 ,0x33 ,0x44 ,0x00 ,0x00);

hattesen, 2023-06-27:
@orgua, it sounds great that you plan to look at supporting the RP2040 in OneWireHub.

Ideally an RP2040 implementation should make use of the RP2040 PIO (an IO coprocessor), which allows implementation of serial protocols with single cycle (@125MHz) precision timing, leaving the application/device implementation free of bit-banging code, timing critical sections and avoiding having to disable interrupts. 

If taking on the PIO is too much for the initial port to RP2040, if you define an API with a sufficiently high abstraction level (addressing and sending bytes), it would be reasonably easy for someone to add a PIO implementation later (via a pull request).

Here are a few inspirational resources I have collected that all implement a 1-wire protocol **master** using the RP2040 PIO for the time-critical serial encoding/decoding:

* [OneWireNg](https://github.com/pstolarz/OneWireNg)
  * driver for RP2040 using PIO: [OneWireNg/src/platform/OneWireNg_PicoRP2040PIO.h](https://github.com/pstolarz/OneWireNg/blob/master/src/platform/OneWireNg_PicoRP2040PIO.h)
  * driver for RP2040 using bit-banging: [OneWireNg/src/platform
/OneWireNg_PicoRP2040.h](https://github.com/pstolarz/OneWireNg/blob/master/src/platform/OneWireNg_PicoRP2040.h)
* [Interfacing 1-Wire devices to the Pico](https://github.com/raspberrypi/pico-examples/tree/eca13acf57916a0bd5961028314006983894fc84/pio/onewire)
* [RP2040-PIO-1-Wire-Master](https://github.com/stefanalt/RP2040-PIO-1-Wire-Master)
* [The Pico In C: A 1-Wire PIO Program](https://www.i-programmer.info/programming/hardware/14527-the-pico-in-c-a-1-wire-pio-program.html)

mr-miky, 2024-06-26:
> Just found the `calibrate_by_bus_timing` script and it seems to work now with `IPL = 25` :)
>
> Unfortunately I still don't understand what the other values mean that were configured in `platform.h` [...]

Can you test this ??

```
#define PIN_TO_BASEREG(pin)             (0)
#define PIN_TO_BITMASK(pin)             (1ul << pin)
#define DIRECT_READ(base, pin)          !!(PIN_TO_BITMASK(pin) & sio_hw->gpio_in)  //digitalRead(pin)
#define DIRECT_WRITE_LOW(base, pin)     sio_hw->gpio_clr = PIN_TO_BITMASK(pin)     // digitalWrite(pin, LOW)
#define DIRECT_WRITE_HIGH(base, pin)    sio_hw->gpio_set = PIN_TO_BITMASK(pin)     //digitalWrite(pin, HIGH)
#define DIRECT_MODE_INPUT(base, pin)    sio_hw->gpio_oe_clr = PIN_TO_BITMASK(pin)  // pinMode(pin,INPUT)
#define DIRECT_MODE_OUTPUT(base, pin)   sio_hw->gpio_oe_set = PIN_TO_BITMASK(pin)  // pinMode(pin,OUTPUT)
using io_reg_t = uint32_t; // define special datatype for register-access
constexpr uint8_t VALUE_IPL {25}; // instructions per loop, uncalibrated so far - see ./examples/debug/calibrate_by_bus_timing for an explanation

```