orgua/OneWireHub #139: Failed communication with master device - works with BUS PIRATE

jejmule, 2024-06-19:
It's not really a bug... let's me explain

I have a real device : CNAP, medical device. The device use CUFF as consumable, the duration of use is stored in a DS2432 chip.
I am working to bypass the memory information to use over-time cuff for formation, tests, and student lesson. CUFF cost approx 1'000$ so we just want to use them for measurement during study.

I have recorded one wire communication with pulse view. I have written a code to emulate the DS2432 using OneWireHub on arduino UNO.
When I test my slave with BUS PIRATE, I can mimic the communication I have recorded using real cuff. But it doesn't work when using the "real" master.

I have recorded the communication with pulse view : 

Below is "original" communication between CNAP instrument and a CUFF
[image]

Below is the "real" master with my emulated device using OneWireHub lib
[image]
My guess is that the slave have a high state between two reicv command and address sent from the master is not visible on the bus. Master tried this several and then display an error message

Below is the BUS PIRATE sending request to the emulated device using OneWireHub
[image]
Here there is a delay between the command and the target address when bus pirate is the master.

I have the feeling, that the issue is coming from timing. The master is sending the target address 80us after sending the command. When BUS PIRATE sent it 844us after the target adress.

Do you have any insight where I can look for to tweak the library behavior to match the master timing?

Thank you,
Jérôme

jejmule, 2024-07-09:
Nobody have a guess...

fmushthafa, 2024-07-11:
Hi Jérôme, I am also working in the similar project (consumable on medical device) but I even haven't been able to read the data. I tried using my rasbpi to read the data directly by using file manager with 1-wire protocol enabled. However, all I got is FF FF FF in every byte and I think I should use another approach (using uC like arduino). 

Anyway, instead of trying to emulate the slave (CUFF) can we just overwrite the counter on the DS2432 so it will read the CUFF as if it's new or unlimited?